nextcloud/nextcloudpi #76: Can't connect to APPS

polmourgos, 2017-07-31:
One more issue i have is I can't connect to apps.I have installed till now 6 times the nextcloudpi to raspberry cose i didnt know how to resolve deferent issues.The last problem i face is the nextcloud do not connect to apps so i can download what i want.I really cant solve this.All the rest works exelent
Thanks for any help again.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-07-31:
Hi,
Can you explain what you mean by "cannot connect to apps"?

do you mean "installing apps from the admin panel of Nextcloud?"

polmourgos, 2017-07-31:
> do you mean "installing apps from the admin panel of Nextcloud?"

nachoparker, 2017-08-01:
Do you have internet access from your RPi? 

I know you set up a static address, so... did you correctly set up the gateway?

Can you `ping ownyourbits.com` from your RPi?

It seems to me like you just didn't configure your network correctly

polmourgos, 2017-08-01:
Yes i have internet and  ping to sites.
When I clik on the apps then a circle appears and stay there spinning and spinning and after some time only some basics apps appears.I really like the nextcloupi img and I really want to have a successfull instalation.But till now I dont have success.  I wont give up.
Thank you nachoparker for your help.

nachoparker, 2017-08-07:
I have never seen that before, seems to me like you have a connectivity issue... can you try the following from the RPi?

```
curl https://apps.nextcloud.com | head -30
```

and paste the output here?

polmourgos, 2017-08-08:
Issue has solved after I did a clear installation to RP3.
I think was router port port forward  problem.

closed 2017-08-08 by polmourgos

nachoparker, 2017-08-08:
good job :)

polmourgos, 2017-08-11:
I haft to say that i reinstall the nextcloudpi and have again the same problem as before.It is something that is not going well.    I gave  the above command and i got this answer,

curl https://apps.nextcloud.com | head -30
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:16 --:--:--     0

<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="utf-8">
    <meta name="viewport" content="width=device-width, initial-scale=1">
    <meta name="description" content="">
    <meta name="author" content="">
    <link rel="shortcut icon" href="/static/assets/img/favicon.png"/>
    <title>All apps - App Store - Nextcloud</title>
    <link rel="stylesheet" href="/static/assets/css/fonts.css">
    <link rel="stylesheet"
          href="/static/vendor/bootstrap/dist/css/bootstrap.min.css">
    <link rel="stylesheet" href="/static/assets/css/style.css">
    <link rel="stylesheet" href="/static/assets/css/theme.css">
    <link rel="alternate" type="application/atom+xml" title="News" href="/en/feeds/releases.atom" />
    <link rel="alternate" type="application/rss+xml" title="News" href="/en/feeds/releases.rss" />
    <script src="/static/vendor/bootstrap.native.min.js" async></script>
    <script src="/static/vendor/element-closest.js" defer></script>
    
    <script defer src="/static/public/app/list.js"></script>

</head>

<body>
<div id="container">
    

<div class="banner">
curl: (23) Failed writing body (0 != 5758)

I am very qurious to discover why this happening.Thanks for any help. 
Pol

nachoparker, 2017-08-11:
I think there's something wrong with your hardware/network :S

polmourgos, 2017-08-18:
`I think there's something wrong with your hardware/network :S`

When i readed your suggestion i couldn't take to serius.But after all my efforts that did not have results i decide to try a friend's router .And this was my problem.After I replaced it everything worked like it sould be.My ZTE router was a disaster.The new TPLink  is life saver.Tooks me so many days tasting and trying diferent thinks.All ports forwording corect,also the https ssl  everything.
I apologize that i din't belied you from the begin.A big thank you Nachoparker.

nachoparker, 2017-08-20:
Nice, I am glad it finally works! :)

It can be hard to spot faulty hardware