nextcloud/nextcloudpi #78: /media external USB drive mappings and location

Shadowstreik, 2017-07-31:
Hi! I'm curious about the external USB mappings. I have automount running for a 500GB external drive and it's attached. The mappings for /media are a hair confusing. Could you clarify for my understanding?

It's now mapped like this...

/media
    |_USBdrive
        |_backups
    |   |_lost+found
    |_USBDrive1
    |_<My USB Drive>
        |_backups
        |_lost+found

I created the backups directory on <My USB Drive>. But, it's also showing up in <USBdrive>. I didn't put that there. It, pretty much, just copied over.

Are USBdrive and USBdrive1 on the actual USB drive or are they on the SD card? If <USBdrive and USBdrive1> is on the SD card, aren't I limited, in total data memory ('data' for example)? Is it necessary for it to copy everything in <MyUSB drive> to <USBdrive>?

I guess, more importantly, will my backups I have generated that are told to be saved at <My USB drive> be only on the USB drive and not on the SD card?

Thank you.

nachoparker, 2017-08-01:
There's an explanation [here](https://ownyourbits.com/2017/05/24/nextcloudpi-gets-automount-backups-nextcloud-online-installation-nextcloud-12-and-more/), and another one github question with more details [here](https://github.com/nextcloud/nextcloudpi/issues/68)

Basically, there will always be a `/media/USBdriveX` folder for each of your drives/partitions and __if__ your drive has a label there will also be a _symbolic link_. It is recommended that you always use the label name ( `/media/mylabel` ).

> Is it necessary for it to copy everything in MyUSB drive to USBdrive? 

Both `/media/USBdrive` and `/media/My USB Drive` are equivalent, different names for the same thing, but there is no data duplication.

> If USBdrive and USBdrive1 is on the SD card...

The files under `/media/X` are not in your SD, but in your external HDD(s). You can check what is mounted where with the `mount` command.

Also, I really recommend that you use a label without spaces... in Linux that is calling for trouble ;) (things tend to not work well)

> I guess, more importantly, will my backups I have generated that are told to be saved at be only on the USB drive and not on the SD card?

They will be saved wherever you tell it to. If you specify a directory in your SD card, such as `/home/pi`, then they will be in SD card. If you specify a directory on your USB drive, such as `/media/USBdrive`, `/media/myDriveLabel` or `/media/USBdrive/ncbackups`, then they will not be in the SD card.

You can even plug in a second hard drive and schedule your backups to be placed there for more redundancy.

Hope it makes sense

closed 2017-08-01 by nachoparker

Shadowstreik, 2017-08-01:
Thank you for the clarification. I scheduled the auto-backup and noticed there isn't a selection for time of day. Could that be added?

nachoparker, 2017-08-01:
It could certainly... when other priorities are cleared.

I suggest you create a feature request with this, to keep issues separate ;)