nextcloud/nextcloudpi #80: Error transferring https://serversite/status.php server unavailable

Ricardosgeral, 2017-08-03:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

V0.18.6

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

Nextcloudpi_07-21-17
### Expected behavior
The server should be accessible in Windows clientand In the browser 

### Actual behaviour
In Windows client Replies error transfering.../status.php
Server is unavailable

In web shows a blue screen with the following info
Please check if data directory contains a file .ocdata in its root

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

Ricardosgeral, 2017-08-03:
Not sure if this error occurred after upgrading to v0.18.6 or during uploading from Windows client via web Davies a .mov file with more than to 2gb
Help please

nachoparker, 2017-08-07:
Your data directory is gone. Did you move it to an external USB?

Ricardosgeral, 2017-08-08:
No I didn't
Only now i have access to the raspberry. And it has strange behaviour. It asks for root password for maintenance...I think i have to reinstall it

If i make a new instalation with another sd card I guess there is no way to save all the data that was transfered to the external usb drive, right?

karnschi, 2017-08-08:
Hey there.
I guess I got the same issue after upgrading to v0.18.6
"Server replied: Internal Server Error"

nachoparker, 2017-08-08:
> Only now i have access to the raspberry. And it has strange behaviour. It asks for root password for maintenance...I think i have to reinstall it

That happens because your hard drive is gone, or at least it is not available on boot as instructed on `/etc/fstab`

The data that was transfered to the USB drive should stay there as long as the USB drive is not broken. If you moved it to `/media/USBdrive/ncdata` for example, look in `/media/USBdrive/ncdata/admin/files` ( for the admin user )

closed 2017-08-08 by nachoparker