nextcloud/nextcloudpi #81: Error setting up letsencrypt

EvdM66, 2017-08-04:
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_07-21-17

### Expected behavior
I am trying to setup Letsencrypt. I think that my port forwarding (443) and the noip configuration work allright. The domain I want to use is cloud.avondel.com which is mapped on evdm66.ddns.net
My DNS contains a corresponding CNAME mapping too.

### Actual behaviour

It fails with the following error message:

[image]

The error references a DNS A/AAAA record. I have a CNAME record. I did try to change it to type A, but that only accepts a fixed IP for destination. Here it gets a little to technical for me unfortunately.

Any suggestion is much appreciated!
</details>

JimmyKater, 2017-08-04:
sure that you have opened and forwarded all neccessary ports for it? 
CNAME shouldn't be any problem... i have set a CNAME record myself and it works beautifully

Shadowstreik, 2017-08-04:
An make sure the no-ip domain is directed to your physical location's IP assigned by your ISP. A helpful tool that could be used is http://www.whatismypublicip.com/, but there are others that can be found via a web search.

EvdM66, 2017-08-04:
Thanks so much!

As far as I can tell the situation is like this:

I have forwarded both 443 ports to my raspberry:

[image]

Whatsmypublicip.com shows this:

[image]

And my noip entry looks like this:

[image]

Ping to both cloud.avondel.com and to evdm66.ddns.net end up at 178.85.89.36, so that seems okay too.

Not sure where to look next...

nachoparker, 2017-08-06:
> I have forwarded both 443 ports to my raspberry:

You need to forward both ports 443 and 80 for the Let's Encrypt challenge to work

closed 2017-08-08 by nachoparker

EvdM66, 2017-08-08:
Thank you very much, thad did solve it indeed!

nachoparker, 2017-08-08:
: )

fab1ton, 2017-08-15:
Hello,

I am facing the same problem as you had it before. In my case I opened already both ports 80 and 443 and it is still not working. I am using ddclient with namecheap.
I actually just reinstalled NextcloudPi. Before everything was working just fine, just now I am facing this issue.

jonaskor, 2017-11-23:
Hey everyone,
i'm was facing the same problem as cryptodigger. Just reinstalled NextcloudPi (after being stuck in maintenance mode) and the letsencrypt plugin was giving me a "fetching timeout" error.
I didn't change the DNS service nor the port forwarding behavior in my router. 
But I noticed, that my public IP differed from the one that pinging my domain gave me.
In the end the problem was, that the DNS service didn't forward to the right domain. After updating the IP at the DNS service provider (in my case changeip.com) letsencrypt now is working fine!
Cheers!

nachoparker, 2017-11-23:
@jonaskor no need to reinstall for that! ask in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for help