nextcloud/nextcloudpi #82: samba smbd.service failed

candide10, 2017-08-05:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.18.6
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
v0.18.5 - downloaded about a week ago (~7/28/17)
### Expected behavior
I'm trying to get a system setup with Nextcloud and samba on an rpi2 Nextcloud box.  On my first try I got samba and nextcloud working but without internet access to Nextcloud and while troubleshooting I managed to break Nextcloud.  Samba was still working though.  Now I've tried again from a fresh install, followed the first basic steps.  Nextcloud is working, but samba won't start.   I get this error:

Launching samba
insserv: warning: current start runlevel(s) (empty) of script 'smbd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevels(s) (0 1 2 3 4 5 6) of script 'smbd' overrides LSB defaults (0 1 6).
Job for smbd. service failed.  See 'systemctl status smbd.service' and 'journalctl -xn' for details.
Added user pi.
Done. Press any key...

I don't know how to copy the text from systemctl and it's a lot to type and I don't know how to open journalctl -xn

Any help would be appreciated.  I'm not sure what to do next.

nachoparker, 2017-08-07:
Did you make any configuration changes in `nextcloudpi-config` or the web interface?

If you access through SSH then you can copy and paste the output of 

`journalctl -xn` and `systemctl status smbd`

nachoparker, 2017-08-07:
In the NextCloudPi_07-21-17 image this didn't happen, but in the NextCloudPi_07-29-17 it does.

I was able to reproduce, there is a segmentation fault.

```
systemctl status smbd           
smbd.service - Samba SMB Daemon                          
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled)                                                          
   Active: failed (Result: core-dump) since Mon 2017-08-07 18:47:42 UTC; 42s ago                                       
     Docs: man:smbd(8)       
           man:samba(7)      
           man:smb.conf(5)   
  Process: 1438 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=dumped, signal=SEGV)                                       
 Main PID: 1438 (code=dumped, signal=SEGV)                 

Aug 07 18:47:38 raspberrypi systemd[1]: Starting Samba SMB Daemon...                                                   
Aug 07 18:47:42 raspberrypi systemd[1]: smbd.service: main process exited, code=dumped, status=11/SEGV                 
Aug 07 18:47:42 raspberrypi systemd[1]: Failed to start Samba SMB Daemon.                                              
Aug 07 18:47:42 raspberrypi systemd[1]: Unit smbd.service entered failed state.
```

Pretty sure it comes from this fix: #48

Clearly, smbclient and smbd are incompatible together with mysqld from Stretch.

Probably will have to wait to the Stretch release to fix it  :(

Now what should we do? which one do we choose, smbclient, or samba?

candide10, 2017-08-08:
Thanks for looking into this!  I couldn't figure out what I was doing wrong.  Ultimately I installed Nextcloudpi from a previous image (6-28-17) and that worked.  Nextcloud and samba are both working for me now.  Now on to the next challenge...

Again, thanks for your work!

nachoparker, 2017-08-08:
I am generating a new image with samba but without `php-smbclient`.

I'll then test if doing `sudo apt-get remove php-smbclient` fixes the issue on current images

nachoparker, 2017-08-08:
image generated, reopened #48

closed 2017-08-08 by nachoparker