nextcloud/nextcloudpi #83: Automatic Update of NoIP DDNS

cottoa00, 2017-08-05:
### QUESTIONS (not an issue... not sure how to tag as such)

Have my NextcloudPi up and running... just received notice from NoIP (my free DDNS service) that I need to reregister (to maintain free service) or upgrade to paid service.
Somewhere in my review of all the forums and docs that you have a script that automatically reregisters my system. Can you please confirm... if not I want to be able to keep my DDNS alive ;-)

Cheers,
Tony Cotton

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

cottoa00, 2017-08-05:
Did some more digging in docs... think the cron job was updating noIP whith any IP change not sending "renewal requests" for domain name maintenace every 30 days.
Went ahead and confirmed my free domain for another month and should just upgrade to paid to avoid the monthly renewal ;-)

closed 2017-08-05 by cottoa00

nachoparker, 2017-08-06:
There is a service (or daemon) called `noip2` that updates the IP address associated with your domain.

The registration of the domain itself, has to be renewed every month unless you upgrade. It's just clicking a link once a month.