nextcloud/nextcloudpi #84: Problems after shutting down router and raspberry pi

JVAQUEROM, 2017-08-05:
Hi,

I went out of my home for a few days and I decided to power off everything, router and raspberry pi. When I have come back, I cannot connect through my no-ip web, and I cannot access my router configuration (127.0.0.1). It says it cannot connect and so on. Accessing nextcloud through local IP works properly.

What can I do?

Thank you in advance for your help!!

JimmyKater, 2017-08-07:
pls make sure that your no-ip account hasn't been disabled... since you need to confirm it (in the free-version) every 30 days.

nachoparker, 2017-08-07:
> EDIT: It seems router config is gone. It looks like I might have pressed the reset button accidentally or something, altough it is weird because you usually need to press it with something sharp...I'll try to reconfigure and tell my trials and tribulations!!

LOL, if that's the case you will need to set up port forwarding again

closed 2017-08-07 by nachoparker