nextcloud/nextcloudpi #85: Img for Odroid xu4 test?

polmourgos, 2017-08-07:
Hi.
After i have done a perfect setup with my pi3 with the nextcloupi img i got the previews nextcloud backup,I am ready to unistall the dietpi from my odroid XU4 and install ( if is posible ) the nextcloudpi.
Even if is in beta verion.
Thanks.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-08-07:
I am not sure of what you mean. Do you want to try a NextCloudPi image for the odroid?

polmourgos, 2017-08-08:
Yes I do.I have already test it but odroid XU4 do not booting.With raspberry the nextcloudpi img is really fast,I i am imaging that it will be double fast with odroid if can work.

nachoparker, 2017-08-08:
Unfortunately, a new NCP version would have to be built on top of an OS supported by the odroid.

I have been looking at [dietpi](http://dietpi.com/), but that is not immediate. It would be a considerable amount of work to port everything to a new OS.

One day I want to try it though.

closed 2017-08-08 by nachoparker