nextcloud/nextcloudpi #86: How do I use NextCloudPi side-by-side with Calibre-server on port 8080?

Magister52, 2017-08-07:
Hi.

I've installed NextCloud manually in the past on my RPi3 and have always been able to run my own personal websites out of Apache2 as well as my own eBook-server running Calibre-server on port 8080. Since configuring NextCloud with all the correct security-settings is a pain I thoughts I'd go for NextCloudPi as a base-setup and then add Calibre and also add my own personal websites to the /var/www/html folder.

However, when I do this, I can only access the Calibre server locally on port 8080 even though my router forwards the port and I can't access any sites I add to the html-folder Apache normally uses. I assume this is a configuration problem that can easily be solved. I would like things to work like this:

Nextcloud-server --> https://mydomain.net
Calibre-server --> https://mydomain.net:8080
My other websites --> https://mydomain.net/subsite (where subsite is any folder added to /var/www/html/)

How do I fix this? I've tried fiddling around with the configuration-files under apache2/sites-available to no avail thinking it's a virtual host issue. I'm also thinking that it might be a SSL issue because I never used SSL in the past. I'm using letsencrypt and a free noip-dns.

I hope I haven't posted this in the wrong place or been to vague in what I am trying to accomplish. I have never used GitHub before.

Sincerely,
Magister52

nachoparker, 2017-08-07:
Hi,

Yes, you have to setup the virtual hosts in `sites-available` the way you want, and then `sudo a2enmod mysite`.

Look at this code for inspiration ( this sets up a different website on port 4443, which is the web interface )

https://github.com/nextcloud/nextcloudpi/blob/master/nextcloudpi.sh#L37

You can replace the location of the certificates to the ones in `/etc/letsencrypt` if you have them

closed 2017-08-07 by nachoparker

Magister52, 2017-08-07:
Thanks for the reply.

While this did help me better understand the issue at hand, I couldn't quite get it to work. I can get it to listen to port 8080 locally but not when I'm using https://mydomain.net:8080 because it drops me a ERR_SSL_PROTOCOL_ERROR immediately.

I am also wondering if I can change the setup so that Nextcloud runs from https://mydomain.net/nextcloud instead of https://mydomain.net and revert to the default Apache-site instead. 

Sincerely,
Magister52

nachoparker, 2017-08-07:
You will have to get some practice with the Apache configuration. Look at examples, and make sure that your directories, SSL parameters and locations are correct.

> I am also wondering if I can change the setup so that Nextcloud runs from https://mydomain.net/nextcloud instead of https://mydomain.net and revert to the default Apache-site instead.

Sure, just configure the virtualhost `DocumentRoot`