nextcloud/nextcloudpi #88: Nextcloud refuses to connect after expanding filesystem

ozzymo00, 2017-08-07:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
0.18.6
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloud_07-29-17
### Expected behavior
Expand filesystem and have all of the 200GB microsd card available
### Actual behaviour
Expand the file system, all space is available but now I can no longer connect to my Nextcloud installation through the local IP.  The connection is refused
### Steps to reproduce, from a freshly installed image
1)  Flash image
2)  Connection to local ip works and NextcloudPi installation is functioning properly
3)  SSH
4)  sudo raspi-config
5)  Advanced options, expand filesystem
6)  sudo reboot
7)  after rebooting try to connect to local ip and get connection refused
8)  SSH still works

### Include logs

df -h 
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       181G  2.0G  172G   2% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           489M     0  489M   0% /dev/shm
tmpfs           489M  6.7M  482M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           489M     0  489M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  52% /boot

lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 183.4G  0 disk 
├─mmcblk0p2 179:2    0 183.3G  0 part /
└─mmcblk0p1 179:1    0  41.7M  0 part /boot

<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-08-07:
that's a fat SD card!

Paste the output of `systemctl status apache2`
Paste the output of `tail -100 /var/log/apache2/*.log`

Paste the output of `systemctl status mysqld`
Paste the output of `tail -100 /var/log/mysql/*.log`

ozzymo00, 2017-08-07:
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-08-07 16:20:39 UTC; 2h 29min ago
  Process: 712 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

tail: option used in invalid context -- 1

mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
   Active: active (running) since Mon 2017-08-07 16:20:56 UTC; 2h 30min ago
 Main PID: 860 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─860 /usr/sbin/mysqld

2017-08-07 15:25:35 1989181440 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-08-07 15:25:35 1989181440 [Note] InnoDB: The InnoDB memory heap is disabled
2017-08-07 15:25:35 1989181440 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-08-07 15:25:35 1989181440 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-08-07 15:25:35 1989181440 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-08-07 15:25:35 1989181440 [Note] InnoDB: Using Linux native AIO
2017-08-07 15:25:35 1989181440 [Note] InnoDB: Using generic crc32 instructions
2017-08-07 15:25:35 1989181440 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-08-07 15:25:36 1989181440 [Note] InnoDB: Completed initialization of buffer pool
2017-08-07 15:25:36 1989181440 [Note] InnoDB: Highest supported file format is Barracuda.
2017-08-07 15:25:38 1989181440 [Note] InnoDB: 128 rollback segment(s) are active.
2017-08-07 15:25:38 1989181440 [Note] InnoDB: Waiting for purge to start
2017-08-07 15:25:38 1989181440 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3494752
2017-08-07 15:25:38 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-07 15:25:38 1989181440 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-07 15:25:38 1989181440 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-07 15:25:38 1989181440 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
2017-08-07 15:40:49 1638634480 [Note] /usr/sbin/mysqld: Normal shutdown

2017-08-07 15:40:49 1638634480 [Note] Event Scheduler: Purging the queue. 0 events
2017-08-07 15:40:49 1426060272 [Note] InnoDB: FTS optimize thread exiting.
2017-08-07 15:40:49 1638634480 [Note] InnoDB: Starting shutdown...
2017-08-07 15:40:51 1638634480 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-08-07 15:40:52 1638634480 [Note] InnoDB: Shutdown completed; log sequence number 3604819
2017-08-07 15:40:52 1638634480 [Note] /usr/sbin/mysqld: Shutdown complete

2017-08-07 15:41:05 1989414912 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-08-07 15:41:05 1989414912 [Note] InnoDB: The InnoDB memory heap is disabled
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-08-07 15:41:05 1989414912 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Using Linux native AIO
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Using generic crc32 instructions
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-08-07 15:41:05 1989414912 [Note] InnoDB: Completed initialization of buffer pool
2017-08-07 15:41:06 1989414912 [Note] InnoDB: Highest supported file format is Barracuda.
2017-08-07 15:41:09 1989414912 [Note] InnoDB: 128 rollback segment(s) are active.
2017-08-07 15:41:09 1989414912 [Note] InnoDB: Waiting for purge to start
2017-08-07 15:41:09 1989414912 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3604819
2017-08-07 15:41:09 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-07 15:41:09 1989414912 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-07 15:41:09 1989414912 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-07 15:41:09 1989414912 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
2017-08-07 15:49:41 1639236592 [Note] /usr/sbin/mysqld: Normal shutdown

2017-08-07 15:49:41 1639236592 [Note] Event Scheduler: Purging the queue. 0 events
2017-08-07 15:49:41 1426060272 [Note] InnoDB: FTS optimize thread exiting.
2017-08-07 15:49:41 1639236592 [Note] InnoDB: Starting shutdown...
2017-08-07 15:49:42 1639236592 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-08-07 15:49:43 1639236592 [Note] InnoDB: Shutdown completed; log sequence number 3644686
2017-08-07 15:49:43 1639236592 [Note] /usr/sbin/mysqld: Shutdown complete

2017-08-07 15:51:00 1988886528 [Note] InnoDB: Using mutexes to ref count buffer pool pages
[... 19 lines cut ...]
2017-08-07 16:20:20 1639236592 [Note] Event Scheduler: Purging the queue. 0 events
2017-08-07 16:20:20 1426060272 [Note] InnoDB: FTS optimize thread exiting.
2017-08-07 16:20:20 1639236592 [Note] InnoDB: Starting shutdown...
2017-08-07 16:20:21 1639236592 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-08-07 16:20:23 1639236592 [Note] InnoDB: Shutdown completed; log sequence number 3678878
[... cut ...]
2017-08-07 16:20:56 1989255168 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3678878
2017-08-07 16:20:56 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-07 16:20:56 1989255168 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-07 16:20:56 1989255168 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-07 16:20:56 1989255168 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

nachoparker, 2017-08-07:
sorry, try

```
tail -n 100 /var/log/apache2/*.log
```

ozzymo00, 2017-08-07:
==> /var/log/apache2/error.log <==
[Mon Aug 07 15:42:40.751176 2017] [authz_host:error] [pid 943:tid 1897042992] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /ncp.js failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:40.764233 2017] [authz_host:error] [pid 943:tid 1871877168] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /loading-small.gif failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:40.776995 2017] [authz_host:error] [pid 943:tid 1888654384] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /ncp.css failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:40.786215 2017] [authz_host:error] [pid 943:tid 1880265776] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /ncp-logo.png failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:40.987635 2017] [authz_host:error] [pid 943:tid 1863488560] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /favicon.png failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:45.308768 2017] [authz_host:error] [pid 943:tid 1855099952] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:45.310282 2017] [:error] [pid 943:tid 1855099952] [client 192.168.1.102] ModSecurity: Warning. Pattern match "^[\\\\d.:]+$" at REQUEST_HEADERS:Host. [file "/etc/modsecurity/modsecurity_crs_21_protocol_anomalies.conf"] [line "98"] [id "960017"] [rev "2"] [msg "Host header is a numeric IP address"] [data "192.168.1.115:4443"] [severity "WARNING"] [ver "OWASP_CRS/2.2.9"] [maturity "9"] [accuracy "9"] [tag "OWASP_CRS/PROTOCOL_VIOLATION/IP_HOST"] [tag "WASCTC/WASC-21"] [tag "OWASP_TOP_10/A7"] [tag "PCI/6.5.10"] [tag "http://technet.microsoft.com/en-us/magazine/2005.01.hackerbasher.aspx"] [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:45.376354 2017] [:error] [pid 943:tid 1855099952] [client 192.168.1.102] ModSecurity: Rule 75ef5a80 [id "981172"][file "/etc/modsecurity/modsecurity_crs_41_sql_injection_attacks.conf"][line "157"] - Execution error - PCRE limits exceeded (-8): (null). [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:45.376906 2017] [:error] [pid 943:tid 1855099952] [client 192.168.1.102] ModSecurity: Rule 75ef5a80 [id "981172"][file "/etc/modsecurity/modsecurity_crs_41_sql_injection_attacks.conf"][line "157"] - Execution error - PCRE limits exceeded (-8): (null). [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:45.377532 2017] [:error] [pid 943:tid 1855099952] [client 192.168.1.102] ModSecurity: Rule 75ef5a80 [id "981172"][file "/etc/modsecurity/modsecurity_crs_41_sql_injection_attacks.conf"][line "157"] - Execution error - PCRE limits exceeded (-8): (null). [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:45.512676 2017] [:error] [pid 943:tid 1855099952] [client 192.168.1.102] ModSecurity: Warning. Match of "streq %{SESSION.CSRF_TOKEN}" against "ARGS:csrf_token" required. [file "/etc/modsecurity/modsecurity_crs_43_csrf_protection.conf"] [line "34"] [id "981144"] [msg "CSRF Attack Detected - Invalid Token."] [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:45.532616 2017] [:error] [pid 943:tid 1618998320] [client 192.168.1.102] ModSecurity: Warning. Operator LT matched 5 at TX:inbound_anomaly_score. [file "/etc/modsecurity/modsecurity_crs_60_correlation.conf"] [line "33"] [id "981203"] [msg "Inbound Anomaly Score (Total Inbound Score: 3, SQLi=1, XSS=0): Host header is a numeric IP address"] [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKdX8AAQEAAAOvpegAABAO"]
[Mon Aug 07 15:42:47.273485 2017] [authz_host:error] [pid 943:tid 1846711344] [client 192.168.1.102:62700] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Mon Aug 07 15:42:47.274983 2017] [:error] [pid 943:tid 1846711344] [client 192.168.1.102] ModSecurity: Warning. Pattern match "^[\\\\d.:]+$" at REQUEST_HEADERS:Host. [file "/etc/modsecurity/modsecurity_crs_21_protocol_anomalies.conf"] [line "98"] [id "960017"] [rev "2"] [msg "Host header is a numeric IP address"] [data "192.168.1.115:4443"] [severity "WARNING"] [ver "OWASP_CRS/2.2.9"] [maturity "9"] [accuracy "9"] [tag "OWASP_CRS/PROTOCOL_VIOLATION/IP_HOST"] [tag "WASCTC/WASC-21"] [tag "OWASP_TOP_10/A7"] [tag "PCI/6.5.10"] [tag "http://technet.microsoft.com/en-us/magazine/2005.01.hackerbasher.aspx"] [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKd38AAQEAAAOvpekAABEP"]
[Mon Aug 07 15:42:47.345164 2017] [:error] [pid 943:tid 1846711344] [client 192.168.1.102] ModSecurity: Rule 75ef5a80 [id "981172"][file "/etc/modsecurity/modsecurity_crs_41_sql_injection_attacks.conf"][line "157"] - Execution error - PCRE limits exceeded (-8): (null). [hostname "192.168.1.115"] [uri "/ncp-launcher.php"] [unique_id "WYiKd38AAQEAAAOvpekAABEP"]
[Mon Aug 07 15:42:47.345715 2017] [:error] [pid 943:tid 1846711344] [client 192.168.1.102] ModSecurity: Rule 75ef5a80 [id "981172"][file "/etc/modsecurity/modsecurity_crs_41_sql_injection_attacks.conf"][line "157"] - Execution error - PCRE limits exceeded (-8): ( [... cut ...]
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:48:18 +0000] "GET / HTTP/1.1" 302 507 "-" "-"
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:48:18 +0000] "GET /HNAP1/ HTTP/1.1" 302 519 "-" "-"
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:48:18 +0000] "POST /JNAP/ HTTP/1.1" 302 517 "-" "-"
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:49:18 +0000] "GET /HNAP1/ HTTP/1.1" 302 519 "-" "-"
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:49:18 +0000] "GET / HTTP/1.1" 302 507 "-" "-"
127.0.1.1:80 192.168.1.1 - - [07/Aug/2017:15:49:18 +0000] "POST /JNAP/ HTTP/1.1" 302 517 "-" "-"

nachoparker, 2017-08-07:
do you have `modsecurity` on?

does it come up if you do `sudo systemctl restart apache2`?

ozzymo00, 2017-08-07:
When I try to restart is using your command above I get "Job for apache2.service failed. See 'systemctl status apache2.service' and 'journalctl -xn' for details."

When I run systemctl status apache2.service this is what I get:

● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-08-07 20:22:10 UTC; 56s ago
  Process: 2173 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

When I run journalctl -xn I get no journal files were found

nachoparker, 2017-08-08:
I am at a loss here... will order a big SD card to see if it affects.

Anybody else seeing this issue?

HrJosch, 2017-08-09:
I seem to be facing exactly the same issue with a 16 GB SD card and struggling since updating to the web interface version. Before I had a perfectly running ncp instance up for three months. 

`==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Tue Aug 08 11:52:45.013672 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: APR compiled version="1.5.1"; loaded version="1.5.2"
[Tue Aug 08 11:52:45.015797 2017] [:warn] [pid 9158:tid 3069881920] ModSecurity: Loaded APR do not match with compiled!
[Tue Aug 08 11:52:45.017301 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: PCRE compiled version="8.35 "; loaded version="8.35 2014-04-04"
[Tue Aug 08 11:52:45.026010 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: LUA compiled version="Lua 5.1"
[Tue Aug 08 11:52:45.027353 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: LIBXML compiled version="2.9.1"
[Tue Aug 08 11:52:45.028813 2017] [:notice] [pid 9158:tid 3069881920] Status engine is currently disabled, enable it by set SecStatusEngine to On.
[Tue Aug 08 11:52:45.845893 2017] [ssl:warn] [pid 9160:tid 3069881920] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:52:46.120388 2017] [mpm_event:notice] [pid 9160:tid 3069881920] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 11:52:46.195509 2017] [core:notice] [pid 9160:tid 3069881920] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 11:56:33.680120 2017] [mpm_event:notice] [pid 9160:tid 3069881920] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 11:56:39.529579 2017] [ssl:warn] [pid 11350:tid 3069308480] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:56:41.288812 2017] [ssl:warn] [pid 11370:tid 3069308480] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:56:42.147093 2017] [mpm_event:notice] [pid 11370:tid 3069308480] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 11:56:42.155660 2017] [core:notice] [pid 11370:tid 3069308480] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 12:17:47.746719 2017] [mpm_event:notice] [pid 11370:tid 3069308480] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 12:18:14.806187 2017] [ssl:warn] [pid 742:tid 1992814592] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:15.134649 2017] [ssl:warn] [pid 900:tid 1992814592] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:16.005581 2017] [mpm_event:notice] [pid 900:tid 1992814592] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 12:18:16.005907 2017] [core:notice] [pid 900:tid 1992814592] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 12:18:15.076233 2017] [ssl:warn] [pid 710:tid 1992142848] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:18.683488 2017] [ssl:warn] [pid 835:tid 1992142848] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:19.005874 2017] [mpm_event:notice] [pid 835:tid 1992142848] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 12:18:19.006273 2017] [core:notice] [pid 835:tid 1992142848] AH00094: Command line: '/usr/sbin/apache2'
[Wed Aug 09 15:19:34.933029 2017] [mpm_event:notice] [pid 835:tid 1992142848] AH00491: caught SIGTERM, shutting down
[Wed Aug 09 17:20:00.863213 2017] [ssl:warn] [pid 824:tid 1992704000] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 09 17:20:01.118411 2017] [ssl:warn] [pid 940:tid 1992704000] AH01909: 2001:16b8:2a:d200:681b:6693:9c98:8c0a:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 09 17:20:02.005663 2017] [mpm_event:notice] [pid 940:tid 1992704000] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Aug 09 17:20:02.005978 2017] [core:notice] [pid 940:tid 1992704000] AH00094: Command line: '/usr/sbin/apache2'
[Wed Aug 09 17:24:42.754791 2017] [mpm_event:notice] [pid 940:tid 1992704000] AH00491: caught SIGTERM, shutting down
[Wed Aug 09 17:24:56.675708 2017] [ssl:warn] [pid 727:tid 1992056832] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 09 17:24:57.121307 2017] [ssl:warn] [pid 927:tid 1992056832] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 09 17:24:58.004913 2017] [mpm_event:notice] [pid 927:tid 1992056832] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Aug 09 17:24:58.005158 2017] [core:notice] [pid 927:tid 1992056832] AH00094: Command line: '/usr/sbin/apache2'
[Wed Aug 09 17:32:32.226218 2017] [mpm_event:notice] [pid 927:tid 1992056832] AH00491: caught SIGTERM, shutting down
[... cut ...]
[Wed Aug 09 21:17:49.388671 2017] [authz_host:error] [pid 1124:tid 1985422384] [client 192.168.111.20:57610] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 09 21:17:51.875431 2017] [mpm_event:notice] [pid 1083:tid 1992880128] AH00491: caught SIGTERM, shutting down`

`systemctl status apache2.service
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Mi 2017-08-09 22:48:42 CEST; 14s ago
  Process: 2670 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)`

Any help would be highly appreciated.

ozzymo00, 2017-08-09:
Yea, it's strange.  I had a previous installation up and running for a few months also.  I then tried the web interface image when it was released and I have never been able to get it working since.  I really don't want to use an external hard drive and would prefer to use a large microsd card.  Hopefully we can get to the bottom of this.

nachoparker, 2017-08-10:
@ozzymo00 What are you getting exactly? a blank page with return code 500? A message of 'internal error'?

@HrJosch Same question for you. How did you update to the web interface version (steps)?

nachoparker, 2017-08-10:
@HrJosch more questions for you... what is your problem exactly? the new web interface doesn't work? or Nextcloud doesn't work either?

Are you using IPv6? (output of `ip address`)

If that's the case, you might need to include

`Require local`

In `/etc/apache2/sites-available/ncp.conf`

And see if the `AH01753` errors go away

HrJosch, 2017-08-10:
Hi @nachoparker, after upgrading to the new webinterface version I faced errors meaning my database is inaccessible. I had no clue how to solve it and assumed it could be a currupt USBdrive. The logs I posted above come from a fresh installation and database with the latest image provided. Both interfaces are inaccessible and i do think the problem is related to apache not starting the service for some reason. Is my assumption that apache is serving both, the web interface and nc correct? 

My ncp.conf looks the following:

`Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile /etc/letsencrypt/live/$  SSLCertificateKeyFile /etc/letsencrypt/li$</VirtualHost>
<Directory /var/www/ncp-web/>
  Require host localhost
  Require ip 127.0.0.1
  Require ip 192.168
  Require ip 10
</Directory>`

Output of ip address:

`1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:72:d6:c3 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::eef2:962f:a781:d2c0/64 scope link tentative 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 74:da:38:61:93:6d brd ff:ff:ff:ff:ff:ff
    inet 192.168.111.17/24 brd 192.168.111.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 2001:16b8:3c:9300:d0bf:bb30:32b:8bf6/64 scope global noprefixroute dynamic 
       valid_lft 6802sec preferred_lft 3202sec
    inet6 fe80::7f5b:f52b:9ee7:7dbf/64 scope link`

nachoparker, 2017-08-10:
> Is my assumption that apache is serving both, the web interface and nc correct

that is correct

You do have ipv6 addresses, probably provided by your router's DHCP. Please, add the `Require local` line, I think you should have access to the ncp-web on 4443

HrJosch, 2017-08-10:
@nachoparker Added the line and rebooted raspi, but still:

`sudo systemctl status apache2.service -l
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Do 2017-08-10 19:25:12 CEST; 23min ago

Aug 10 19:25:12 MyNextCloud apachectl[1288]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
Aug 10 19:25:12 MyNextCloud apachectl[1288]: SSLCertificateFile: file '/etc/letsencrypt/live/mynextcloud.mydomain.com/fullchain.pem' does not exist or is empty
Aug 10 19:25:12 MyNextCloud apachectl[1288]: Action 'start' failed.
Aug 10 19:25:12 MyNextCloud apachectl[1288]: The Apache error log may have more information.
Aug 10 19:25:12 MyNextCloud systemd[1]: apache2.service: control process exited, code=exited status=1
Aug 10 19:25:12 MyNextCloud systemd[1]: Failed to start The Apache HTTP Server.
Aug 10 19:25:12 MyNextCloud systemd[1]: Unit apache2.service entered failed state.`

I exchanged my domain with a synonyme, hope that‘s fine.

What needs to happen with /etc/letsencrypt/live/mynextcloud.mydomain.com/fullchain.pem?

HrJosch, 2017-08-10:
Or is there any way to assign a regular ipv4 address to my node?

nachoparker, 2017-08-10:
> What needs to happen with /etc/letsencrypt/live/mynextcloud.mydomain.com/fullchain.pem?

this is a different one, did you change that line? does the file `/etc/letsencrypt/live/mynextcloud.mydomain.com/fullchain.pem` exist?

HrJosch, 2017-08-10:
It is not existing as I couldn‘t enroll letsencrypt certificates, due to my domain not being reachable during enrollment.  I did not change that line.

nachoparker, 2017-08-10:
aaaaaaaaah that's new information

then change those lines back to 

```
SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem                                                  
SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key
```
That issue was fixed in `v0.19.1`

HrJosch, 2017-08-10:
That was the key, thank you very much for your kind hint, @nachoparker :)
After deploying your changes in ncp.conf and nextcloud.conf,  apache was willing to start again:

`sudo systemctl status apache2.service -l
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: active (running) since Do 2017-08-10 21:05:28 CEST; 3min 13s ago
  Process: 1348 ExecStop=/usr/sbin/apachectl stop (code=exited, status=0/SUCCESS)
  Process: 1355 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 1359 (apache2)
   CGroup: /system.slice/apache2.service
           ├─1359 /usr/sbin/apache2 -k start
           ├─1363 /usr/sbin/apache2 -k start
           └─1364 /usr/sbin/apache2 -k start

Aug 10 21:05:27 MyNextCloud apachectl[1355]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Aug 10 21:05:28 MyNextCloud systemd[1]: Started The Apache HTTP Server.` 

May I ask, should MyNextcloud.MyDomain.com be set as Hostname instead of MyNextcloud?

The internal WebInterface is up again and I can reach the NC.Webinterface via its internal IP address.

Thank you so much for your effort you put into this fabulous project! I appreciate it a lot and learn from it every day.

HrJosch, 2017-08-10:
I am now struggling to reach my instance via fqdn. Portforwarding is active, no-ip is set up and dnsmasq added its strings. Is this related to the SSL certificates?

nachoparker, 2017-08-10:
great!

If noip DDNS is set up and the ports are open, you should be able to access, with a warning if the SSL is not from `letsencrypt`

Let's keep things separated. Please browse old questions and open another one if you are still struggling

HrJosch, 2017-08-10:
All fine now. Safari wouldn't  recognize the changes made. FF showed me the warning and I was able to enroll my certificates. All is working fine now and I will sync data from scratch :) Thank you!

nachoparker, 2017-08-10:
:D

nachoparker, 2017-08-15:
I'll close now. Reopen if anyone else runs into this

closed 2017-08-15 by nachoparker