nextcloud/nextcloudpi #89: Port 80 is not available (blocked by my isp)

Avelec, 2017-08-07:
Hi,

Thank you for all your hard work. 

My problem is accessing my nexcloud from outside my home network. Trying to setup No-IP and was able to open port 443 in my router but not port 80 because it's not available (blocked). Is there away to change from listen to port 80 to other port that I can open in my router. Thanks in advanced!

nachoparker, 2017-08-07:
you only need port 80 for letsencrypt. If it is not accessible, you will have to manually do the DNS challenge method.

For the rest, you don't need it.

It's funny that they allow HTTPS port but not HTTP port   

dO_ob

closed 2017-08-07 by nachoparker

Avelec, 2017-08-07:
Thanks for the reply. I tried to activate no-ip from nextcloudpi-config and this is what I got:

Launching no-ip

Auto configuration for Linux client of no-ip.com.

Can't gethostbyname for dynupdate.no-ip.com
Network must be operational to create configfile. Ending!
System config value trusted_domains => 3 set to string cloudstorage.ddns.net
System config value overwrite.cli.url set to string https://cloudstorage.ddns.net
Done. Press any key...

And still cannot access from the noip domain. Thanks.

PS: Can I use Dataplicity?

nachoparker, 2017-08-07:
> Can't gethostbyname for dynupdate.no-ip.com

Do you have internet connectivity from the Pi?

You can use whatever you want, but it won't be supported. I personally try to avoid using cloud services such as Dataplicity, more so if they have control over your Pi. The whole purpose of the website is to 'own' 'your' 'bits' 

:P

Avelec, 2017-08-07:
OK, rebooted the pi and tried setup no-ip again and it wasn't give me the error : Can't gethostbyname for dynupdate.no-ip.com. I was able to connect from my phone with 4G data, but has to allowed untrusted website to continued. So how can I "manually do the DNS challenge method" like you mentioned in the above post. So I can use let'e encript. Thanks.

nachoparker, 2017-08-07:
Nice!

Unfortunately I have not tried that method myself. It sounds a bit advanced.

[Here](https://certbot.eff.org/docs/using.html) is the manual

If you can't make it work you can accept the security exception in all your devices. Before letsencrypt existed that was our only option ;)

Avelec, 2017-08-07:
Thanks, I'll try it and post back if got it working. 

That's is the problem with att uverse fiber (modem/router device). I'm currently using router behind router setup to get my ipcam working from outside my home.