nextcloud/nextcloudpi #90: sync speed issues using raspberry pi 3

tommyguuuun, 2017-08-07:
Hi,
today i installed the latest version (v0.18.6) on my raspberry pi 3. I connected my ext usb drive to the system and didn't really change anything since then, except some theming options and adding users. Both systems, the computer and raspberry pi 3 are connected trough cable to a fritzbox.

When i try to sync my files (about 100GB and 19000 Files), the transferspeed is unbelievably slow (about 15 B/s up to 18 KB/s). Also, after a short while the web interface isn't available anymore and the nextcloud sync tool says "operation aborted" - I had a "normal" nextcloud installation before and it didn't happen there. Everything was normal.

Could someone help me? I don't have much knowledge about linux systems and i'm really happy to have found something like nextcloudpi and really want to use it.

nachoparker, 2017-08-07:
well, that's not normal. I get around 11 Mibps with a RPi2 and an ooold USB drive.

Are you using a cheap or old power supply?

tommyguuuun, 2017-08-07:
i mean 15 BYTES not Mibps ;). I don't really know which power supply, it came with a prebuild package... but the ext usb hdd has it's one powersupply.

nachoparker, 2017-08-07:
If you could test with another one... cheap power supplies are the number one reason for low performance and hardware unstability

tommyguuuun, 2017-08-07:
nope, i tried to different other supplies and always the same reaction. After rebooting i have speeds of up to 2 Mbit/s for about 2 minutes and then i get the error "service unavailable" and after it the speed slows down dramatically...

nachoparker, 2017-08-07:
have you tried `top` or `uptime` when the system is in that state?

tommyguuuun, 2017-08-07:
yes, apache is always on top, followed by php7-fpm and the cpu usage is at about 50% overall.

uptime gets me:
 23:38:42 up 16 min,  1 user,  load average: 5,36, 5,24, 3,53

it went ok by uploading some bigger videos, but then it crashed again. Also i'm not able to higher the max upload filesize... could this be a reason?

nachoparker, 2017-08-07:
your load is really really high. Are you using an external hard drive? what SD card are you using?

Max upload size is 2GB. It has been improved in NC 13

tommyguuuun, 2017-08-07:
yes, i'm using an external hard drive. As SD card i use a Sandisk Ultra SDHC 16GB.

nachoparker, 2017-08-07:
Your CPU is not _that_ high. Try `iotop`. I think that your HDD might be the bottleneck

tommyguuuun, 2017-08-07:
command not found..

nachoparker, 2017-08-07:
install it xD

tommyguuuun, 2017-08-07:
1175 be/4 root        0.00 B/s    0.00 B/s  0.00 % 99.99 % [ext4lazyinit]
 1173 be/3 root        0.00 B/s    0.00 B/s  0.00 % 99.99 % [jbd2/sda1-8]
  889 be/4 www-data    2.97 M/s    2.97 M/s  0.00 % 95.77 % php-fpm: pool www
 1512 be/4 www-data    3.36 M/s    3.36 M/s  0.00 % 95.08 % php-fpm: pool www
 1236 be/4 www-data    3.36 M/s    3.36 M/s  0.00 % 94.06 % php-fpm: pool www
 5838 be/4 root        0.00 B/s    0.00 B/s  0.00 % 88.53 % [kworker/u8:0]
17036 be/4 www-data    3.36 M/s    0.00 B/s  0.00 % 78.19 % php-fpm: pool www
   93 be/3 root        0.00 B/s    0.00 B/s  0.00 %  4.41 % [jbd2/mmcblk0p2-]
 5048 be/4 mysql      10.08 K/s   77.26 K/s  0.00 %  4.02 % mysqld
 1273 be/4 mysql       6.72 K/s   20.16 K/s  0.00 %  0.90 % mysqld
 1110 be/4 mysql       0.00 B/s  241.87 K/s  0.00 %  0.76 % mysqld
  883 be/4 mysql       0.00 B/s    0.00 B/s  0.00 %  0.46 % mysqld
 1682 be/4 www-data    0.00 B/s    2.01 M/s  0.00 %  0.00 % php-fpm: pool www
 1024 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
    1 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % init
    2 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kthreadd]
    3 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [ksoftirqd/0]
 1028 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
    5 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kworker/0:0H]
 1030 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
    7 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [rcu_sched]
    8 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [rcu_bh]
    9 rt/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [migration/0]
   10 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [lru-add-drain]
   11 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [cpuhp/0]
   12 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [cpuhp/1]
   13 rt/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [migration/1]
   14 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [ksoftirqd/1]
  527 be/4 nobody      0.00 B/s    0.00 B/s  0.00 %  0.00 % thd --daemon --triggers /etc/t~--user nobody /dev/input/event*
   16 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kworker/1:0H]
   17 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [cpuhp/2]
   18 rt/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [migration/2]
   19 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [ksoftirqd/2]
   21 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kworker/2:0H]
   22 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [cpuhp/3]
   23 rt/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [migration/3]
   24 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [ksoftirqd/3]
 1049 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
   26 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kworker/3:0H]
   27 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kdevtmpfs]
   28 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [netns]
   29 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [khungtaskd]
   30 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [oom_reaper]
   31 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [writeback]
   32 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kcompactd0]
   33 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [crypto]
   34 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [bioset]
   35 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kblockd]
   36 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [watchdogd]
   38 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [rpciod]
   39 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [xprtiod]
   40 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kswapd0]
   41 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [vmstat]
   42 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [nfsiod]
 1031 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
 7213 be/4 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kworker/1:1]
 1117 be/4 mysql       0.00 B/s    0.00 B/s  0.00 %  0.00 % mysqld
 1032 be/4 www-data    0.00 B/s    0.00 B/s  0.00 %  0.00 % apache2 -k start
   52 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [kthrotld]
   53 be/0 root        0.00 B/s    0.00 B/s  0.00 %  0.00 % [bioset]

tommyguuuun, 2017-08-07:
i think it could have something to do with the max upload size... it still tries to upload a file, which is 2,57GB in size... how can i increase or workaround this? but don't really know, if this is the problem.

tommyguuuun, 2017-08-07:
well... i'm sick of it. I will try to do a complete, fresh installation. What would be the best way to prepare my external hdd? completely format it before? to which file system? or just erase the partitions?

nachoparker, 2017-08-07:
The upload will cut at 2GB, and it won't complete

Your USB drive is ext4... should be fine as it is formatted. If you have another HDD to test if the problem is the USB drive... that would be the best

I don't think you have to start from scratch.

tommyguuuun, 2017-08-07:
sadfully i don't have any other drive... but it worked with my old installation one week ago without any problems

nachoparker, 2017-08-08:
what OS did your old installation have? also Raspbian?

tommyguuuun, 2017-08-08:
yes, also raspbian. I did a complete clean install and format and still the same problem. I think i figured out, that the problem are the 2G. It syncs well, till it hits a file over 2G, than after a while it looses its connection and gets an error. Then it tries to resync the file and fails again. I delted one file by hand and it worked ok after it, but the current file i'm not able to delete on the server, so it tries to download it every time i try to synch

nachoparker, 2017-08-08:
Well, like I said it is **impossible** that a file over 2GB is going to be transfered until NC 13. 

I don't know how that affects the sync client, but it should not affect the web interface

You have other methods for uploading your big files. Look at this blog post

https://ownyourbits.com/2017/04/18/different-ways-to-access-your-nextcloud-files/

closed 2017-08-08 by nachoparker