nextcloud/nextcloudpi #91: Updating n-update will update nextcloud to v12.0.1

Ricardosgeral, 2017-08-08:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

Ricardosgeral, 2017-08-08:
Please delete this issue

closed 2017-08-08 by Ricardosgeral

nachoparker, 2017-08-08:
duplicate of #92