nextcloud/nextcloudpi #92: nc-update updates nextcloud to v12.0.1?

Ricardosgeral, 2017-08-08:
Does it? 
I tried to update nextcloud but it gives error in first step about expected files
.Well-known

nachoparker, 2017-08-08:
`nc-update` does not update NextCloud, only NextCloudPi ( the extras, the web interface... )

To upgrade Nextcloud you have to use the built in 'updater' (admin page)

I still have to test the upgrade from 12.0.0. I'll get back to you.

`.well-known` is created during the letsencrypt authentication process... you can just delete it from `/var/www/nextcloud` and try again.

Like I said.. I still haven't tested the update myself

closed 2017-08-08 by nachoparker

Shadowstreik, 2017-08-08:
> 
> Check for write permissions
> The following places can not be written to:
> 
>     /var/www/nextcloud/updater/../.htaccess
> 
> 

Does this fix that issue?

> chown -R www-data:www-data /var/www/nextcloud

This is the result of using the Updater within Nextcloud, "Stable" update channel.

Thank you.

Shadowstreik, 2017-08-08:
Posted issue to the support area @ Nextcloud forums.

nachoparker, 2017-08-08:
This is an old story. They still are not clear on what the permissions should be, and they keep changing the recommended permissions.

I already reported it when it happened with NC11 [here](https://github.com/nextcloud/server/issues/4509)

I'll fix it now for NextCloudPi. Please, everyone update to `v0.19.1` and try again

I'll also cleanup `.well-known` after `letsencrypt` is set up

Shadowstreik, 2017-08-08:
> -------- Original Message --------
> Local Time: August 8, 2017 5:21 PM
> UTC Time: August 8, 2017 9:21 PM

Updated NCP to 0.19.1, was able to run the updater within NC and performed a reboot for good measure. All is well.

Thank you for updating NCP to compensate. It's very much appreciated!

--
CONFIDENTIALITY NOTE: Some E-Mail communication may occur using a secure method. The information transmitted, including attachments,
is intended only for the person(s) or entity to which it is addressed and
may contain confidential and/or privileged material. Any review, re-transmission,
dissemination or other use of, or taking of any action in reliance upon this
information by persons or entities other than the intended recipient is prohibited.
If you received this in error, please contact the sender via E-Mail, only,
and destroy any copies of this information. IMPORTANT: This E-Mail is intended
for the use of the individual addressee(s) named above and may contain information
that is confidential, privileged or unsuitable for overly sensitive persons with
low self-esteem, no sense of humor or irrational religious and/or political beliefs.
Any dissemination, distribution or copying of this E-Mail is not authorized
(either explicitly or implicitly) and constitutes and irritating social faux-pas.
Also, any E-Mail sent from the originating author claims ownership of all E-Mail
sent and received on local computer(s) in addition to all non-solicited E-Mails
received. CAUTION: Unless the word 'absquatulation' has been used in its correct
context somewhere other than in this warning, it does not have any legal nor
grammatical use and may be ignored. No animals were harmed in the transmission of
this E-Mail, although the kelpie next door is living on borrowed time, let me tell
you. Those of you with an overwhelming fear of the unknown will be gratified to
learn that there is no hidden message revealed by reading this warning backwards,
so just ignore that alert notice from Microsoft. ANTIDOTE: However, by pouring a
complete circle of salt around yourself and your computer you can ensure that
no harm befalls you and your pets. If you have received this E-Mail in error,
please add some nutmeg and egg whites, whisk and place in a warm oven for 40 minutes.

Sent with [ProtonMail](https://protonmail.com) Secure Email.

nachoparker, 2017-08-09:
@Shadowstreik  great, thanks for the feedback :D

Ricardosgeral, 2017-08-09:
I did the update to the nextcloud v12.0.1 after your update thanks. Everything seems OK. No issues found so far

Avelec, 2017-08-09:
Thanks for the excellent support. After update is done which should I pick:

Keep maintenance mode active?
Yes (for usage with command line tool)  No (for usage of the web based updater)

nachoparker, 2017-08-09:
Pick "no"

;)