nextcloud/nextcloudpi #93: QUESTION: NC-DATADIR action behavior

Shadowstreik, 2017-08-09:
Hello. I have all of my content built and it currently resides all on a 64Gb SD card along with NextCloud and NCP. I run and store backups on an external hard drive.

In regards to NC-DATADIR, it reads, "Change your data dir to a new location, like a USB drive". Very straight-forward. My question is regarding pre-existing data stored on the SD card...

Will pre-existing SD card-stored data remain on the SD card and remain viewable to the originating user and folks it has been shared to while anything new will be stored at the external path identified in NC-DATADIR?

Will pre-existing SD card data currently on the SD card be migrated to the external drive path entered into NC-DATADIR and remain viewable to the originating user and folks it has been shared to while anything new will be stored at the external path identified in NC-DATADIR?

nachoparker, 2017-08-10:
Hi,

The exact behaviour right now is 
- the data is copied from the current data dir to the new location
- the old data is left there (originally `/var/www/nextcloud/data`, but not viewable through Nextcloud
- all changes will only be applied to the new location

You can see the code [here](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-datadir.sh#L64)

closed 2017-08-10 by nachoparker

Shadowstreik, 2017-08-10:
Ok, thank you. That is very helpful.

- Shadowstreik

nachoparker, 2017-08-10:
anytime ;)