nextcloud/nextcloudpi #94: Best way to move big files/folders to nexctcloudpi USB drive from a linux/windows pc

Avelec, 2017-08-09:
I have 2TB HDD setup with nextcloudpi. Since the 2GB limit, I haven't been able to copy/upload my pictures  (total of 1.5 TB). I have tried plugin the 2TB HDD to a laptop running Linux mint, when trying to open the "ncdata folder" it gave me error: 
"The folder content cannot be displayed"
"You do not have permission necessary to view the content of ncdata"

nachoparker, 2017-08-10:
You will have to do it as superuser, because the data folder is owned by `www-data` user (the HTTP server user) on the Pi.

If you do it from a file manager, for example, `nemo`, type `sudo nemo`. If you do it from the terminal, do `sudo cp...`

You can also use SCP, NFS or SMB for big files so you do not have to unplug the USB each time. See [this article](https://ownyourbits.com/2017/04/18/different-ways-to-access-your-nextcloud-files/)

closed 2017-08-10 by nachoparker

Avelec, 2017-08-10:
Hello Nachoparker, thank you for your always prompt response.

I tried to activate samba in  sudo nextcloudpi-config, but got this error:

Launching samba
Job for smbd.service failed. See 'systemctl status smbd.service' and 'journalctl -xn' for details.
Done. Press any key...
I have usb hdd attached. The samba user name&pw  is the same as the one I use with putty to ssh into the Rpi, correct?
Thanks in advance.

nachoparker, 2017-08-10:
If you are using the NextCloudPi_07-29-17 image, there was this bug #82 

Probably if you type `systemctl status smbd` you will get

```
systemctl status smbd           
smbd.service - Samba SMB Daemon                          
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled)                                                          
   Active: failed (Result: core-dump) since Mon 2017-08-07 18:47:42 UTC; 42s ago                                       
     Docs: man:smbd(8)       
           man:samba(7)      
           man:smb.conf(5)   
  Process: 1438 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=dumped, signal=SEGV)                                       
 Main PID: 1438 (code=dumped, signal=SEGV)                 

Aug 07 18:47:38 raspberrypi systemd[1]: Starting Samba SMB Daemon...                                                   
Aug 07 18:47:42 raspberrypi systemd[1]: smbd.service: main process exited, code=dumped, status=11/SEGV                 
Aug 07 18:47:42 raspberrypi systemd[1]: Failed to start Samba SMB Daemon.                                              
Aug 07 18:47:42 raspberrypi systemd[1]: Unit smbd.service entered failed state.
```

Avelec, 2017-08-10:
Yes, I'm using the NextCloudPi_07-29-17.img. Yes this is what I got:

● smbd.service - Samba SMB Daemon
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled)
   Active: failed (Result: core-dump) since Thu 2017-08-10 14:14:01 EDT; 25min a                                                                                                                               go
     Docs: man:smbd(8)
           man:samba(7)
           man:smb.conf(5)
  Process: 23528 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=dumped, signal=SEGV                                                                                                                               )
 Main PID: 23528 (code=dumped, signal=SEGV)

Aug 10 14:14:01 raspberrypi systemd[1]: Starting Samba SMB Daemon...
Aug 10 14:14:01 raspberrypi systemd[1]: smbd.service: main process exited, c...V
Aug 10 14:14:01 raspberrypi systemd[1]: Failed to start Samba SMB Daemon.
Aug 10 14:14:01 raspberrypi systemd[1]: Unit smbd.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

nachoparker, 2017-08-10:
I am sorry to say that you have to use the latest image due to that bug.

I recommend that you use `nc-backup` first, so you can then `nc-restore` your Nextcloud instance.

Avelec, 2017-08-10:
But  I had already update to 12.0.1 2 days ago using through the updater app in the admin page.

I'm doing the update again this time through nc-nextcloud and nc-init. Then I ran nc-restore and got the error:
Launching nc-restore
INFO: overwriting old instance
restore files...
rm: cannot remove ‘/var/www/nextcloud/data/.opcache/850b9065d108158eb20bb00ad0892f94/var/www/nextcloud’: Directory not empty
mv: inter-device move failed: ‘/media/USBdrive/nextcloud-bkp_20170810.tar-tmp/nextcloud’ to ‘/var/www/nextcloud’; unable to remove target: Directory not empty
restore database...
grep: /var/www/nextcloud/config/config.php: No such file or directory
Error reading data directory
Done. Press any key...

and when I tried to log in to nexcloud I got:
[image]

nachoparker, 2017-08-10:
I am afraid you are going to have to download image NextCloudPi_08-08-17 and start from there. Upgrading is not enough, it doesn't matter admin page, or `nc-nextcloud`

:(

Avelec, 2017-08-10:
OK, ran the nc-restore the 2nd time seems to fixed it. I can log back in to nextcloud but the not showing any of the files/folders from the HDD.

Avelec, 2017-08-10:
Launching nc-scan, the result:

Starting scan for user 1 out of 1 (admin)
Exception during scan: "files/e8482945c9011a1fad7f0293d9760c26" is locked
#0 /var/www/nextcloud/lib/private/Files/Storage/Common.php(704): OC\Lock\DBLocki                                                                                                                               ngProvider->acquireLock('files/e8482945c...', 2)
#1 /var/www/nextcloud/lib/private/Files/Storage/Wrapper/Wrapper.php(588): OC\Fil                                                                                                                               es\Storage\Common->acquireLock('scanner::', 2, Object(OC\Lock\DBLockingProvider)                                                                                                                               )
#2 /var/www/nextcloud/lib/private/Files/Cache/Scanner.php(318): OC\Files\Storage                                                                                                                               \Wrapper\Wrapper->acquireLock('scanner::', 2, Object(OC\Lock\DBLockingProvider))
#3 /var/www/nextcloud/lib/private/Files/Utils/Scanner.php(242): OC\Files\Cache\S                                                                                                                               canner->scan('', true, 3)
#4 /var/www/nextcloud/apps/files/lib/Command/Scan.php(161): OC\Files\Utils\Scann                                                                                                                               er->scan('/admin')
#5 /var/www/nextcloud/apps/files/lib/Command/Scan.php(234): OCA\Files\Command\Sc                                                                                                                               an->scanFiles('admin', '/admin', false, Object(Symfony\Component\Console\Output\                                                                                                                               ConsoleOutput), false)
#6 /var/www/nextcloud/3rdparty/symfony/console/Command/Command.php(256): OCA\Fil                                                                                                                               es\Command\Scan->execute(Object(Symfony\Component\Console\Input\ArgvInput), Obje                                                                                                                               ct(Symfony\Component\Console\Output\ConsoleOutput))
#7 /var/www/nextcloud/core/Command/Base.php(161): Symfony\Component\Console\Comm                                                                                                                               and\Command->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfo                                                                                                                               ny\Component\Console\Output\ConsoleOutput))
#8 /var/www/nextcloud/3rdparty/symfony/console/Application.php(818): OC\Core\Com                                                                                                                               mand\Base->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony                                                                                                                               \Component\Console\Output\ConsoleOutput))
#9 /var/www/nextcloud/3rdparty/symfony/console/Application.php(186): Symfony\Com                                                                                                                               ponent\Console\Application->doRunCommand(Object(OCA\Files\Command\Scan), Object(                                                                                                                               Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Out                                                                                                                               put\ConsoleOutput))
#10 /var/www/nextcloud/3rdparty/symfony/console/Application.php(117): Symfony\Co                                                                                                                               mponent\Console\Application->doRun(Object(Symfony\Component\Console\Input\ArgvIn                                                                                                                               put), Object(Symfony\Component\Console\Output\ConsoleOutput))
#11 /var/www/nextcloud/lib/private/Console/Application.php(170): Symfony\Compone                                                                                                                               nt\Console\Application->run(Object(Symfony\Component\Console\Input\ArgvInput), O                                                                                                                               bject(Symfony\Component\Console\Output\ConsoleOutput))
#12 /var/www/nextcloud/console.php(100): OC\Console\Application->run()
#13 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#14 {main}

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 0       | 0     | 00:00:00     |
+---------+-------+--------------+

Log in to nextcloud still not showing any files from the hdd.

nachoparker, 2017-08-10:
Did you move your data directory? you should really start from the latest image

Avelec, 2017-08-10:
When I first installed it, I followed your instructions and did the following:
Enable HDD the nc-automount feature. Then set your Nextcloud data folder in the drive using nc-datadir

Before updating I did nc-backup, after updating I ran  nc-restore. 

Note: I did not "include Data" when backing up, I assumed it's just for backing up all the files I put in the HDD. Will this cause the problem for the files in HDD not showing up anymore?

nachoparker, 2017-08-10:
If you didn't include the data, then your Nextcloud data will not be in the backup.

If you moved the datadir they should still be there, wherever you put them

Avelec, 2017-08-10:
Oops..I made a stupid mistake. So how can I get all the files/folders in the HDD to show up again in nextcloud? I guess all my files/folders should still be in the HDD, right?

nachoparker, 2017-08-10:
right. At the location where you instructed to move them ( `/media/USBdrive/ncdata` by default )

Avelec, 2017-08-10:
So to get the files to show up should I run nc-datadir or nc-database? What is the different between the 2?

nachoparker, 2017-08-10:
`nc-datadir`

`nc-database` is for the database, not for the files. The database is used for listing the files, and also users, calendars... all the rest

Avelec, 2017-08-10:
Did ran the nc-datadir and leave the default location ( /media/USBdrive/ncdata by default ).

Then launching nc-scan, after 3 minutes it found 3 folders which is correct but no files?
It's still not showing any files/folders in nextcloud.

Starting scan for user 1 out of 1 (admin)

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 3       | 0     | 00:03:53     |
+---------+-------+--------------+
Done. Press any key...

Avelec, 2017-08-11:
Tried to run nc-datadir again and got this:

Launching nc-datadir
/media/USBdrive/ncdata is not empty
Done. Press any key...

nachoparker, 2017-08-11:
you have to change it manually. Please try

```
sudo nano /var/www/nextcloud/config/config.php
```

, and modify `datadirectory` to 

```
'datadirectory' => '/media/USBdrive/ncdata',
```

( `ctrl-x` to save )

Avelec, 2017-08-11:
I got it working again, thank you very much for your help. 

Inside the /config.php, I see that it's  'version' => '12.0.1.5'. I think I better just gave up samba, since it's causing some issue.

nachoparker, 2017-08-11:
> I got it working again, thank you very much for your help.
great

> I think I better just gave up samba, since it's causing some issue.
It's up to you, but it's fixed in the new image.

Thanks for the feedback, because I had not noticed this

Avelec, 2017-08-11:
I did a backup image of my working sdcard. I would like to try to activate samba again. 

When you said "fixed in the new image", is that meant I need to download the new image and burned it onto an sd card and start from new again? The problem I was facing yesterday was when I updated to the new version through sudo nextcloudpi-config. Thanks.