nextcloud/nextcloudpi #96: Wrong file size of folders and files

Ricardosgeral, 2017-08-10:
There is a  issue with file sizes of all files shown I  web and clients (Android and Windows at least) 

I've found a solution in git hub that is very simple. It is just a matter of removing a small portion of text in configuration.php

Ricardosgeral, 2017-08-10:
In fileinfo.php. Remove "(1)" 
See https://github.com/nextcloud/server/issues/5031

However I am afraid that this my have side effects. 

I've done it in the  past and solved it but later I hade the status.Php file missing, and I don't know if they are related. 

What should I do to fix the wrong file sizes
Thanks

Ricardosgeral, 2017-08-12:
Sorry the patch is to remove (int) from Getsize function and not (1)

Anyone reported side effects?

nachoparker, 2017-08-14:
Thanks for the info. 

Regarding to NCP, we will wait until it is [rolled out in NC12.0.2](https://github.com/nextcloud/server/issues/5031#issuecomment-320890965)

closed 2017-08-14 by nachoparker