nextcloud/nextcloudpi #97: handle DHCP & DNS so an enduser just has to turn off DHCP on the router

StuartIanNaylor, 2017-08-10:
With DNSMasq I think for endusers its easier for it to handle DHCP & DNS so an enduser just has to turn off DHCP on the router which could simplify things for use.

nachoparker, 2017-08-10:
Thanks for the suggestion. It would indeed make things easier, turn off DHCP in the router vs setup DNS on each device.

Anyone else, thoughts?

testbird, 2017-09-17:
Most routers I know run a dns proxy like dnsmask, and dhcp clients automatically get a DNS name, i.e. raspberrypi.local-network-domain-configured-in-router. The dhcp client of the raspberrypi writes this domain into /etc/resolv.conf, where it can be read for adjusting the list trusted domains for nextcloud.

The *.local domain is another option.
Some routers manage the domain .local (raspberrypi.local), but it should actually be used for peer2peer based mDNS, which the avahi-daemon package that is present on the raspberrypi is supposed to handle. Unfortunately, the avahi/Bonjour/zeroconf/mDNS configuration did not seem to work for me.

Mentioning these options to access the device in the docs (and auto-configuring them as trusted domains in NC) might do quite well.

Keep in mind that with DHCP and DNS on the NAS, every outage or reboot of it could mess up the DHCP and break the network and DNS lookup in the entire LAN. It may be an option for some, but generally I think DHCP and DNS is best handled by the gateway/router.

nachoparker, 2017-09-18:
Hi,

More brainstorming! cool, we need more eyes looking at this ;)

> The dhcp client of the raspberrypi writes this domain into /etc/resolv.conf, where it can be read for adjusting the list trusted domains for nextcloud.

I don't understand why would you want to add the DNS provider in `/etc/resolv.conf` to trusted domains of NC. Trusted domains are for matching the URL that you type in your browser to access NC, and that is never going to be the router/DNS provider. Would you type 8.8.8.8 to access your NC? I'm confused with that suggestion.

> The *.local domain is another option.
Some routers manage the domain .local (raspberrypi.local), but it should actually be used for peer2peer based mDNS, which the avahi-daemon package that is present on the raspberrypi is supposed to handle. Unfortunately, the avahi/Bonjour/zeroconf/mDNS configuration did not seem to work for me.

What's the intention behind this? Make the router's DNS provider aware of the raspi domain for NC? That would be interesting even though depends on router mDNS support

> Keep in mind that with DHCP and DNS on the NAS, every outage or reboot of it could mess up the DHCP and break the network and DNS lookup in the entire LAN. It may be an option for some, but generally I think DHCP and DNS is best handled by the gateway/router.

I aggree with DHCP, but I guess it doesn't hurt to have the option. It is too much to ask for many users have them manually configure every device to point to a hardcoded DNS.

With DNS it is not an issue, because you only set your primary DNS to your NCP and if it's down or if you are outside of your house it will automatically, instantly and painlessly fall back to the second provider.

Also, none of this is needed if you router supports NAT loopback

Ofc, NCP is supposed to be always up and running ;)

nachoparker, 2017-09-18:
Reading your link from the other issue

> http://openmediavault.local (if your router's DHCP server is not broken then http://openmediavault will also work).

Is this based on mDNS like you just said?

testbird, 2017-09-19:
>> http://openmediavault.local (if your router's DHCP server is not broken then http://openmediavault will also work).

> Is this based on mDNS like you just said?

Yes, the .local domain is reserved for mDNS (p2p multicast method for devices to establish DNS names in the LAN in a p2p fashion without the need for central DNS server or router). Exception: some legacy routers used the ".local" DNS domain (centrally) for resolving the names of the current DHCP client leases.

Normally, routers should use another domain chosen by the user (like ".myname.lan" or defaulting to something like ".linkthis", ".ciskco", ".router" ) for resolving the names from the DHCP client leases locally.

All this makes a devices reachable as "devicename.myname.lan", or short "devicename".)

In nextcloud you would not use the "nameserver" string from the resolv.conf to construct a trusted domain, but the "search" string (the local domain provided by the dhcp server).

---
Instead of configuring a "primary dns" for the local router, it is often also possible to configure a local dns entry in the router (for a global domain name) that points lan devices to a local machine/dhcp lease (if router has no NAT loopback).

nachoparker, 2017-09-20:
> Most routers I know run a dns proxy like dnsmask, and dhcp clients automatically get a DNS name, i.e. raspberrypi.local-network-domain-configured-in-router. The dhcp client of the raspberrypi writes [...]

I am not familiar with this option, as my router doesn't seem to have it available. If you could write a little guide about it (here in a comment) I will paste it to the wiki with credit to you. Thanks!

> Instead of configuring a "primary dns" for the local router, it is often also possible to configure a local dns entry in the router (for a global domain name) that points lan devices to a local machine/dhcp lease instead of the public IP (if the router has no automatic NAT loopback).

I will write this option in the wiki, thanks

---

About mDNS and .local domains... I have been playing with it a little bit and it's cool. Definitely would make it easy for many users because you don't have to find the IP, you can access straightaway.

The problem though is that you need the mdns plugin for the glibc resolver installed and configured in `/etc/nsswitch.conf`. This seems to come out of the box in Ubuntu, but not in other minimal systems like Arch (what I use, I had to install and configure the plugin). 

Also, it seems like windows doesn't have mDNS support unless you install some msi (from iTunes!).

I don't know, thoughts on this? It seems like accessing through `https://nextcloudpi.local` will basically only work in Ubuntu ( I haven't researched if Debian as well ), so it might actually make the docs bigger and more complicated with 'little benefit'.

Thoughts?

testbird, 2017-09-20:
For the user manual I think it should not matter how the hostname gets resolved to an IP (mDNS,  central .local domain, or another lan domain . It should all boil down to something like this:

```
The default hostname is "raspberrypi". Your initial nextcloud installation should be readily accessible under https://raspberrypi, otherwise you need to find out the its IP address through one of the following ways.
```
----

I don't own the newest equipment, but I see mDNS is widely used for network browsing/discovery by printers, nas, phones etc... Seems even win10 got updated with a dns-sd service lib, by now, the mDNS service discovery is definitely standard, and the way to go.
https://stackoverflow.com/questions/23624525/standard-mdns-service-on-windows

testbird, 2017-09-20:
For nextcloud to work, the local domain from the resolv.conf needs to be added to the list of trusted domains, of course

testbird, 2017-09-20:
The reason arch has disabled mDNS by default.
https://security.stackexchange.com/questions/39122/possible-exploits-through-avahi-daemon
I think simply mentioning that mDNS/zeroconf/Bonjour service discovery is enabled, and a way to disable it, should be ok for all.

nachoparker, 2017-09-20:
> For nextcloud to work, the local domain from the resolv.conf needs to be added to the list of trusted domains, of course

easy

> win10 got updated with a dns-sd service lib,

I saw that, but I think only for printers... but it's making its way

> By now, I think simply mentioning that the mDNS/zeroconf/Bonjour service discovery is enabled, and mentioning a way to disable it, should be ok for all.

It is already enabled in Raspbian, so no changes there

nachoparker, 2017-09-21:
I implemented it, but it will only come in the next image. When I release it I'll update the wiki, thanks

https://nextcloudpi.local FTW xD

testbird, 2017-09-21:
Thank you, and you are right, using
> https://nextcloudpi.local 

in writing is better: Noticed a webbed viewer first sending http://nextcloudpi to its search surveillance provider and ask the user to click again if they really wanted to open the URL. (This can hardly be considered to be a browser anymore.)

testbird, 2017-09-26:
edited above text for the docs

nachoparker, 2017-09-26:
great thanks, will update the wiki after the release ;)

nachoparker, 2017-09-26:
I'll add the 'help wanted' tag in case somebody wants to configure dnsmasq to act as a dhcp server as well. Should be easy for many people as it's a common configuration pattern.

nachoparker, 2018-02-16:
given the fact that we now support mDNS and many routers support configuring an external DNS provider in the LAN (which would be NCP), I think it would be better to just support those options, and not risk leaving the whole house without internet if the pi breaks or crashes.

closed 2018-02-16 by nachoparker