nextcloud/nextcloudpi #98: Unable to connect to nextcloud web interface after a reboot.

ObfuscateTheWeb, 2017-08-10:
### What version of NextCloudPi are you using? 

12.0.1

### What is the base image that you installed on the SD card? 

NextCloudPi 12.0.1

### Expected behavior

Connects to the web interface as usual. My pi's IP address is 192.168.1.127

### Actual behaviour

Doesn't connect after a reboot. Changed only the data directory to an external hard drive.

### Steps to reproduce, from a freshly installed image

1. Do the basic setup (flash it onto sd card, boot up ,change password , etc)
2. Changed data directory with nextcloudpi-config (via web interface)
3. Tested it by uploading big files. Uploads fine.
4. Did sudo poweroff -h
5. Power on and no longer connects.

### Include logs
<details>
<summary>nextcloud logs</summary>
{"reqId":"WYxfNX8AAQEAAAOnGr4AAAEW","level":4,"time":"2017-08-10T13:33:38+00:00","remoteAddr":"192.168.1.112","user":"user1","app":"webdav","method":"PUT","url":"\/remote.php\/webdav\/Breaking%20Bad\/Season%204\/Breaking%20Bad%20-%20S04E06%20-%20Cornered.m4v","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 388336949 got 5398528\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #8)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('Breaking Bad - ...', Resource id #8)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('Breaking Bad\\\/Se...', Resource id #8, NULL)\\n#3 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v1\\\/webdav.php(71): Sabre\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":151}","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WYxnq38AAQEAAAOoCPYAAFEM","level":4,"time":"2017-08-10T14:10:15+00:00","remoteAddr":"192.168.1.112","user":"user1","app":"webdav","method":"PUT","url":"\/remote.php\/webdav\/Breaking%20Bad\/Season%202\/Breaking%20Bad%20-%20S02E03%20-%20Bit%20by%20a%20Dead%20Bee.m4v","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 386475946 got 107798528\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #8)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('Breaking Bad - ...', Resource id #8)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('Breaking Bad\\\/Se...', Resource id #8, NULL)\\n#3 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v1\\\/webdav.php(71): Sabre\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":151}","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WYxtc38AAQEAAAOoCksAAEID","level":2,"time":"2017-08-10T14:28:03+00:00","remoteAddr":"192.168.1.230","user":"--","app":"core","method":"POST","url":"\/index.php\/login?redirect_url=\/index.php\/apps\/files\/%3Fdir%3D\/%26fileid%3D2","message":"Login failed: 'admin' (Remote IP: '192.168.1.230')","userAgent":"Mozilla\/5.0 (Android 6.0.1; Mobile; rv:55.0) Gecko\/55.0 Firefox\/55.0","version":"12.0.1.5"}
root@raspberrypi:/media/myCloudDrive/ncdata# reeset
bash: reeset: command not found
root@raspberrypi:/media/myCloudDrive/ncdata# reset
root@raspberrypi:/media/myCloudDrive/ncdata# cat newlog
{"reqId":"WYxfNX8AAQEAAAOnGr4AAAEW","level":4,"time":"2017-08-10T13:33:38+00:00","remoteAddr":"192.168.1.112","user":"user1","app":"webdav","method":"PUT","url":"\/remote.php\/webdav\/Breaking%20Bad\/Season%204\/Breaking%20Bad%20-%20S04E06%20-%20Cornered.m4v","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 388336949 got 5398528\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #8)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('Breaking Bad - ...', Resource id #8)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('Breaking Bad\\\/Se...', Resource id #8, NULL)\\n#3 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/event\\\/lib\\\/EventEmitterTrait.php(105): call_user_func_array(Array, Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(479): Sabre\\\\Event\\\\EventEmitter->emit('method:PUT', Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(254): Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v1\\\/webdav.php(71): Sabre\\\\DAV\\\\Server->exec()\\n#8 \\\/var\\\/www\\\/nextcloud\\\/remote.php(162): require_once('\\\/var\\\/www\\\/nextcl...')\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/File.php\",\"Line\":151}","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WYxnq38AAQEAAAOoCPYAAFEM","level":4,"time":"2017-08-10T14:10:15+00:00","remoteAddr":"192.168.1.112","user":"user1","app":"webdav","method":"PUT","url":"\/remote.php\/webdav\/Breaking%20Bad\/Season%202\/Breaking%20Bad%20-%20S02E03%20-%20Bit%20by%20a%20Dead%20Bee.m4v","message":"Exception: {\"Exception\":\"Sabre\\\\DAV\\\\Exception\\\\BadRequest\",\"Message\":\"expected filesize 386475946 got 107798528\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Directory.php(151): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\File->put(Resource id #8)\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/Server.php(1096): OCA\\\\DAV\\\\Connector\\\\Sabre\\\\Directory->createFile('Breaking Bad - ...', Resource id #8)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/sabre\\\/dav\\\/lib\\\/DAV\\\/CorePlugin.php(525): Sabre\\\\DAV\\\\Server->createFile('Breaking Bad\\\/Se...', Resource id #8, NULL)\\n#3 [internal function]: Sabre\\\\DAV\\\\CorePlugin->httpPut(Object(Sabre\\\\HTTP\\\\ [... cut ...]
2017-08-10 16:46:20 1989115904 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 9030249
2017-08-10 16:46:20 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-10 16:46:20 1989115904 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-10 16:46:20 1989115904 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-10 16:46:20 1989115904 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
</details>

If you need more info, just ask..

nachoparker, 2017-08-10:
Hi

> EDIT: I also expanded the filesystem. Using a 16 GB microSD. Reading other issues I saw issue #88
Is my case here similar to that

Probably a duplicate of #88. What do you mean 'no longer connects'? SSH? web?

ObfuscateTheWeb, 2017-08-10:
By web, ssh still works.

nachoparker, 2017-08-10:
Did you try `letsencrypt` and it failed?

Look at https://github.com/nextcloud/nextcloudpi/issues/88#issuecomment-321627480

HrJosch, 2017-08-10:
Same issue I faced `AH00526: Syntax error on line 5`

nachoparker, 2017-08-10:
Yes, that's what it looks like. Please, manually fix as in #88 and upgrade to latest version with `sudo ncp-update`

closed 2017-08-10 by nachoparker

ObfuscateTheWeb, 2017-08-10:
Yes ! It works after making your changes !

nachoparker, 2017-08-10:
> Yes, I tried letsencrypt and it doesn't work.

Need to have ports 80 and 443 open. See #22 #43 #81

ObfuscateTheWeb, 2017-08-10:
But my connection is still "not secure"

ObfuscateTheWeb, 2017-08-10:
[image]

The part where I drew over is my no ip domain

ObfuscateTheWeb, 2017-08-11:
Okay, everything is working now.

I am able to access it on the outside and https with green pad lock is present.

Thanks @nachoparker  and @HrJosch