nextcloud/nextcloudpi #99: Webinterfaces not accessible directly after fresh install

PartTimeDataScientist, 2017-08-10:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.19.1

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_08-08-17

### Expected behavior
Be able to access NextCloud Interface or Management Interface after boot and resizing of SD-card

### Actual behaviour
Neither NextCloud Interface at IP:443 or Managment Interface at IP:4443 are accessible
Also after disabling forceHTTPS IP:80 doesnt work

### Steps to reproduce, from a freshly installed image
1.) Write image to SDHC card
2.) Boot, wait for partition resizing and reboot
3.) Try to connect to web interface

Could be related to #88 and #98, although the error messages suggest different origins. I am also using a 16GB Samsung SDHC card. I have checked the images md5 and verified the write with etcher from windows.

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
/var/www/nextcloud/data/nextcloud.log => file is empty
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: active (running) since Wed 2017-08-09 20:17:28 UTC; 22h ago
  Process: 810 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 879 (apache2)
   CGroup: /system.slice/apache2.service
           ├─879 /usr/sbin/apache2 -k start
           ├─882 /usr/sbin/apache2 -k start
           └─883 /usr/sbin/apache2 -k start

Paste the output of `tail -n 100 /var/log/apache2/*.log`

==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Tue Aug 08 09:40:44.234976 2017] [core:notice] [pid 3655:tid 3069538832] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 10:39:00.640988 2017] [mpm_event:notice] [pid 3655:tid 3069538832] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 10:43:04.812768 2017] [mpm_event:notice] [pid 728:tid 3069591104] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 10:43:05.004244 2017] [core:notice] [pid 728:tid 3069591104] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 10:47:28.819745 2017] [mpm_event:notice] [pid 728:tid 3069591104] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 10:51:49.043264 2017] [mpm_event:notice] [pid 684:tid 3069574720] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 10:51:49.081749 2017] [core:notice] [pid 684:tid 3069574720] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 10:54:11.768740 2017] [mpm_event:notice] [pid 684:tid 3069574720] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 10:58:16.136305 2017] [mpm_event:notice] [pid 681:tid 3069918784] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 10:58:16.156406 2017] [core:notice] [pid 681:tid 3069918784] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 11:52:41.926066 2017] [mpm_event:notice] [pid 681:tid 3069918784] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 11:52:44.724166 2017] [ssl:warn] [pid 9158:tid 3069881920] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:52:45.011076 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity for Apache/2.8.0 (http://www.modsecurity.org/) configured.
[Tue Aug 08 11:52:45.013672 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: APR compiled version="1.5.1"; loaded version="1.5.2"
[Tue Aug 08 11:52:45.015797 2017] [:warn] [pid 9158:tid 3069881920] ModSecurity: Loaded APR do not match with compiled!
[Tue Aug 08 11:52:45.017301 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: PCRE compiled version="8.35 "; loaded version="8.35 2014-04-04"
[Tue Aug 08 11:52:45.026010 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: LUA compiled version="Lua 5.1"
[Tue Aug 08 11:52:45.027353 2017] [:notice] [pid 9158:tid 3069881920] ModSecurity: LIBXML compiled version="2.9.1"
[Tue Aug 08 11:52:45.028813 2017] [:notice] [pid 9158:tid 3069881920] Status engine is currently disabled, enable it by set SecStatusEngine to On.
[Tue Aug 08 11:52:45.845893 2017] [ssl:warn] [pid 9160:tid 3069881920] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:52:46.120388 2017] [mpm_event:notice] [pid 9160:tid 3069881920] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 11:52:46.195509 2017] [core:notice] [pid 9160:tid 3069881920] AH00094: Command line: '/usr/sbin/apache2'
[... 214 lines cut ...]
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-08-09 20:17:27 3062667200 [Note] InnoDB: The InnoDB memory heap is disabled
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-08-09 20:17:27 3062667200 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Using Linux native AIO
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Using generic crc32 instructions
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-08-09 20:17:27 3062667200 [Note] InnoDB: Completed initialization of buffer pool
2017-08-09 20:17:28 3062667200 [Note] InnoDB: Highest supported file format is Barracuda.
2017-08-09 20:17:28 3062667200 [Note] InnoDB: The log sequence numbers 2736199 and 2736199 in ibdata files do not match the log sequence number 2779670 in the ib_logfiles!
2017-08-09 20:17:28 3062667200 [Note] InnoDB: Restoring possible half-written data pages from the doublewrite buffer...
2017-08-09 20:17:30 3062667200 [Note] InnoDB: 128 rollback segment(s) are active.
2017-08-09 20:17:30 3062667200 [Note] InnoDB: Waiting for purge to start
2017-08-10 18:53:58 3062667200 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 2779670
2017-08-10 18:54:00 2533356528 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-10 18:54:00 3062667200 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-10 18:54:00 3062667200 [Note] Recovering after a crash using tc.log
2017-08-10 18:54:00 3062667200 [Note] Starting crash recovery...
2017-08-10 18:54:00 3062667200 [Note] Crash recovery finished.
2017-08-10 18:54:01 3062667200 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-10 18:54:01 3062667200 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
```
</details>

nachoparker, 2017-08-10:
hi,

can you confirm that it works right before resizing the partition and that it stops working right after?

PartTimeDataScientist, 2017-08-10:
Hi,

unfortunately I can not confirm that!

On the first boot from SD-card I immediately see the message "Resizing root filesystem" and the system is rebooting automatically. After the second (first full boot to login prompt) I cannot access anything over the web. The only option for remote access is SSH (after local activation on the shell)

To confirm I just repeated the process and here are the only entries in /var/log/apache2/error.log
```
[Tue Aug 08 11:56:41.288812 2017] [ssl:warn] [pid 11370:tid 3069308480] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 11:56:42.147093 2017] [mpm_event:notice] [pid 11370:tid 3069308480] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 11:56:42.155660 2017] [core:notice] [pid 11370:tid 3069308480] AH00094: Command line: '/usr/sbin/apache2'
[Tue Aug 08 12:17:47.746719 2017] [mpm_event:notice] [pid 11370:tid 3069308480] AH00491: caught SIGTERM, shutting down
[Tue Aug 08 12:18:35.314837 2017] [ssl:warn] [pid 698:tid 3069861440] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:37.126900 2017] [ssl:warn] [pid 799:tid 3069861440] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Aug 08 12:18:38.066352 2017] [mpm_event:notice] [pid 799:tid 3069861440] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Tue Aug 08 12:18:38.066858 2017] [core:notice] [pid 799:tid 3069861440] AH00094: Command line: '/usr/sbin/apache2'
[Thu Aug 10 21:17:33.625328 2017] [proxy:error] [pid 826:tid 3042124848] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/php7.0-fpm.sock (*) failed
[Thu Aug 10 21:17:33.625896 2017] [proxy_fcgi:error] [pid 826:tid 3042124848] [client 192.168.178.21:49991] AH01079: failed to make connection to backend: httpd-UDS
```

Could there be an issue with the php installation? '/run/php/php7.0-fpm.sock' does indeed not exist on the machine and `systemctl status php7.0-fpm.service` shows php-fpm as inactive and a restart doesn't work either.
```
pi@raspberrypi:~ $ systemctl status php7.0-fpm.service
● php7.0-fpm.service - The PHP 7.0 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.0-fpm.service; enabled)
   Active: failed (Result: signal) since Thu 2017-08-10 21:35:17 UTC; 13s ago
     Docs: man:php-fpm7.0(8)
  Process: 1548 ExecStart=/usr/sbin/php-fpm7.0 --nodaemonize --fpm-config /etc/php/7.0/fpm/php-fpm.conf (code=killed, signal=SEGV)
 Main PID: 1548 (code=killed, signal=SEGV)
pi@raspberrypi:~ $ sudo systemctl restart php7.0-fpm.service
Job for php7.0-fpm.service failed. See 'systemctl status php7.0-fpm.service' and 'journalctl -xn' for details.
pi@raspberrypi:~ $ systemctl status php7.0-fpm.service
● php7.0-fpm.service - The PHP 7.0 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.0-fpm.service; enabled)
   Active: failed (Result: signal) since Thu 2017-08-10 21:40:23 UTC; 8s ago
     Docs: man:php-fpm7.0(8)
  Process: 1072 ExecStart=/usr/sbin/php-fpm7.0 --nodaemonize --fpm-config /etc/php/7.0/fpm/php-fpm.conf (code=killed, signal=SEGV)
 Main PID: 1072 (code=killed, signal=SEGV)
pi@raspberrypi:~ $ journalctl -xn
No journal files were found.
```

Hope that helps...

nachoparker, 2017-08-10:
that's weird... why would it die with SEGV? do you have another SD card / power unit to try see if it's a hardware issue?

PartTimeDataScientist, 2017-08-10:
that was actually the first thing I did when it didn't work immediately, although the RPi was running fine on that PSU for two years.

Anyway, I just shut down my SqueezeboxServer to test with a just recently purchased official power supply for the RPi3 and a different SanDisk 8GB microSDHC instead of my two Samsung 16GB SDHC but unfortunately with the same result...

Would it be possible to start from an older image and see if this works - possibly even after an upgrade to the most recent version?

nachoparker, 2017-08-10:
You can try this one: NextCloudPi_07-21-07

magnet:?xt=urn:btih:90cab791f53aa2d36adcb7f1fec1d4b0ca36d79e&dn=NextCloudPi%5F07-21-17&tr=udp%3A%2F%2Ftracker.opentrackr.org%3A1337%2Fannounce

I don't think it will make a difference though :S

PartTimeDataScientist, 2017-08-14:
Tried the old version with the same results.

However, switching from my RasPi2 to the recently purchased RasPi3 everything works as expected with the most recent image (same PSU, same SDHC card, same everything). So it seems to be a RasPi2 related problem, either generic or tied to my specific unit. Unfortunately I am unable to investigate any further at this point as I have no more RasPi2 units for testing available.

nachoparker, 2017-08-14:
Hi,

Thanks for the info. It does seem hardware related.

I have 2 Pi1B, 1 Pi2 and 4 Pi3. Actually, my personal NextCloudPi runs on a RPi2, so I know there is no specific issue with that version. There is even somebody that made it run on a zero.

I think your Pi2 died :S

closed 2017-08-15 by nachoparker