nextcloud/nextcloudpi #100: Is it safe to have nextcloud exposed to the internet?

ObfuscateTheWeb, 2017-08-11:
So i've got everything set up and running fine, including letsencrypt.

Since I port forwarded 80 and 443. I am now able to access my nextcloud domain on the internet via no.ip.com 

I wonder how secure is this? I'm worried that someone might crack into it and then is able to access and modify my files?

I guess this is a side effect of using letsencrypt? Because letsencrypt can't work without port forwarding..

How do I make sure my (now) exposed raspberry pi holding my data is secure?

nachoparker, 2017-08-11:
Hi,

That's a very legitimate concern. Port 80 is only needed for the first setup of `letsencrypt`. You can close that port now, even though NCP redirects automatically to HTTPS.

You need to have 443 ( the HTTPS port ) open in order to access from outside, not only to setup `letsencrypt`. If you want to access from outside, you have to tell your router, simply that.

Now... how secure is this? as much as I can, even though all reviews are welcome. I really recommend that you activate `modsecurity`. 

There are two problems with `modsecurity`
 - At this point it makes `ncp-web` stop working ( the web interface in 4443 )
 - It works with the basic apps, but it is possible that you install an untested app and `modsecurity` breaks it.
In any case, it can be switched on and off from the web interface and `nextcloudpi-config`.

There are many security features in place, both provided by NCP and by Nextcloud. You can learn more in this post

https://ownyourbits.com/2017/03/25/nextcloud-a-security-analysis/

closed 2017-08-11 by nachoparker

ObfuscateTheWeb, 2017-08-11:
I used both.
Modsecurity isn't causing issues with my ncp-web and I can access it just fine?

How do I reset the fail2ban counters?  I mean specifically the total failed and total banned.

nachoparker, 2017-08-11:
> Modsecurity isn't causing issues with my ncp-web and I can access it just fine?

I haven't adapted it to ncp-web yet, I think when you hit 'run' it fails, IIRC

> Isn't port 80 needed to complete the challenge from letsencrypt?

Only the first one I think, not the renewal ( I think xD ).

I have port 80 closed at home myself