nextcloud/nextcloudpi #103: Whitepage after changing database dir, even when putting it back on SD card

Windyo, 2017-08-11:
Fresh install, v 0.19.11, with the latest image from the website.

Setup is a Pi3 with two 3TB HDDs connected to it in Raid1.
I was having random failures to connect to nextcloud - I suspected i twas because of the rapidly-growing database, and migrated it to /mnt/raid0.

NExtcloud then went all blank on me.
Tried migrating it back to the SD card, stayed blank.

I tried to chmod the /ncdatabase/ folder hoping it would allow mariadb to do it's thing, but that failed.

MariaDB error:

mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
  Drop-In: /run/systemd/generator/mysql.service.d
           └─50-mariadb-$database.conf
   Active: failed (Result: exit-code) since Fri 2017-08-11 13:37:25 UTC; 5s ago
  Process: 2682 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 2584 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 2580 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 2577 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 2682 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Aug 11 13:37:25 nextcloud systemd[1]: Starting MariaDB database server...
Aug 11 13:37:25 nextcloud mysqld[2682]: 2017-08-11 13:37:25 1989058560 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 2682 ...
Aug 11 13:37:25 nextcloud mysqld[2682]: 2017-08-11 13:37:25 1989058560 [Warning] Can't create test file /home/pi/ncdatabase/nextcloud.lower-test
Aug 11 13:37:25 nextcloud mysqld[2682]: [95B blob data]
Aug 11 13:37:25 nextcloud mysqld[2682]: 2017-08-11 13:37:25 1989058560 [ERROR] Aborting
Aug 11 13:37:25 nextcloud systemd[1]: mariadb.service: main process exited, code=exited, status=1/FAILURE
Aug 11 13:37:25 nextcloud systemd[1]: Failed to start MariaDB database server.

Windyo, 2017-08-11:
I can make it start with sudo -u mysql mysqld_safe but nextcloud tells me it's in maintenance mode.

Windyo, 2017-08-11:
just in case someone has the same bug:
it seems the automated database migration tool doesn't finish up it's tasks properly, resulting in this.
Solved it by:
- putting the database on an external disk
- stopping apache, mariaDB
- running mariaDB in secure mode to make sure  it can connect the the new db
- reboot
- run sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off

was fixed.

nachoparker, 2017-08-11:
Hey,

Thanks for the feedback

I am looking at the code again and I just realized that if the `mv` command fails it can indeed be left in inconsistent (maintenance mode, and so on). I'll fix it so it fails more gracefully, but chances are the dbdir will be left in unconsistent state if only some of the files are moved.

It is well known that the USB drive needs to be very reliable if we want to move the database there. For instance, if it goes to standby the DB can crash. That's a HW issue and depends on each case.

Could you clarify this? So I can investigate and fix it if needed.

---

On a side note: I was thinking about doing RAID1 with NCP, but I decided that throughput would be terrible (without trying it), so I went for the "autobackup" solution.

I am interested though to see how it performs. Maybe you can post your findings in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

It would be great, to see if we take that route with NCP

nachoparker, 2017-08-11:
> the USB drives are 3.5", connected to a standalone powersupply - can't get much more reliable than that. ;)

That's pretty good, but still I  got reports of standalone powersupply USB drives going into standby mode and causing issues. Mines are pretty old and never got an issue with them ;)

> The errors also appeared when migrating the database from SD card to SD card, even though the card itself is new and passes sanity checks.

> don't know exactly why, but I know it bugs 100% of the time on my end - even though there's nothing particularly novel about my setup (fresh image and all - I just added OpenVPN but it isn't even connected).

That puzzles me. I'll test again and it would be great if you could test things a little bit, if you are up for it. After I test it myself, I mean. I want to know if there's something to fix.

---

> Regarding Raid1 > What are you looking for ? I'm not exactly a computer whiz but I can look into it if you like. FYI it's a software raid, running pretty much verbatim as described here:
https://www.stewright.me/2017/08/create-raid-volume-raspberry-pi/

Yeah, I have been looking into `mdadm`, but also into `btrfs` RAID 1. `btrfs` RAIDS have interesting capabilities, such as balancing, scrubbing, incremental snapshots...

In either case, I was mostly interested in the feasibility of it... I suspect really terrible performance on the Pi. If you are up for trying it, we can open another thread ;)

Windyo, 2017-08-11:
>  That's pretty good, but still I got reports of standalone powersupply USB drives going into standby mode and causing issues. Mines are pretty old and never got an issue with them ;)

They don't even go into standby mode when not plugged in to the PI... They're kind of stupif that way.

>  if you could test things a little bit, if you are up for it
Sure, I'll just back up my instance first... It's pretty new so it's not a big problem.

>  I suspect really terrible performance on the Pi. If you are up for trying it, we can open another thread ;)
Sure. Worst case scenario we prove it's shit and I go back to normal drives + autobackup. Best case we have a bit of fun and I learn more about the linux env. In both cases I win.... I'll let you open the thread.

nachoparker, 2017-08-14:
Implemented in `v0.20.2`

closed 2017-08-14 by nachoparker