nextcloud/nextcloudpi #104: Web panel doesn't allow multiple domain name in letsencrypt interface

martignoni, 2017-08-11:
NextCloudPi version: v0.19.11
Base image: NextCloudPi_08-08-17.img

### Actual behaviour

When trying to generate a letsencrypt certificate with multiple domains, web panel doesn't allow multiple domains entry. E.g. my.cloud.me,my.cloud.net,my.cloud.org. However, the CLI interface allows it without problem

### Expected behavior

Allow multiple domain entry in DOMAIN field of web panel (same features in web panel as in CLI).

### Steps to reproduce, from a freshly installed image

1. Browse to web panel, using a browser, e.g. http://192.168.1.222:4443
1. Click on "letsencrypt" in left column
1. Type "my.cloud.me,my.cloud.net,my.cloud.org" in the DOMAIN field
1. Click on "Run" button
1. Observe that an error is displayed

nachoparker, 2017-08-14:
Hi,
Yes, I was adding protections to the web, so commas are not allowed to prevent code injection.

Maybe it would be safe to add them though

nachoparker, 2017-08-14:
Implemented in `v0.20.1`

closed 2017-08-14 by nachoparker