nextcloud/nextcloudpi #105: ncp-update

senecasphaedra, 2017-08-11:
Will upgrading the nextcloudpi image (i.e., nap-update) maintain all your nextcloud settings?  I would like to update but do not want to lose any of the nextcloud settings already configured.  Thanks for an excellent product.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-08-11:
Upgrading through `sudo ncp-update` or `nc-update` in web panel or in `nextcloudpi-config` only upgrades NextCloudPi stuff, meaning

- NextCloudPi additions:  remote updater system, custom MOTD... those things
- `nextcloudpi-config` launcher
- ncp-web: the web control panel

**It will not modify your Nextcloud instance in any way**

Somewhat related to question #92

So, go ahead! I release soon and then introduce little fixes and improvements almost everyday, like a rolling release. It is recommended to update often to get the latest stuff

;)

closed 2017-08-11 by nachoparker

senecasphaedra, 2017-08-11:
Awesome -- thank you kindly for the very quick response on this.  Much appreciated, and thanks again for the great work on this project.