nextcloud/nextcloudpi #106: Nextcloud seems to use up a lot of ressources when syncing big files

Windyo, 2017-08-11:
So as said before, Nextcloud is running off of a Raid1 3TB array.

When I try to fill it up, nextcloud seems to complain somewhat.

main symptom is the fact that Apache fails to load, complaining about resources, and the client can't sync anymore.
So in general: nextcloud stops working regularly while complaining about resources, and I don't know why.

``top`` show no abnormal use, and ``df`` shows all storage mediusm still ahve place available.

nachoparker, 2017-08-11:
There is a limitation in the max filesize of 2GB. See #94  and #90

You will have to upload those through samba/nfs/ssh

Windyo, 2017-08-11:
I saw that 😉 I’m speaking about multiple 500mb files, for example. Nextcloud fails, goes into hibernation, starts up again afer a while.

nachoparker, 2017-08-11:
Ah, ok

Well, I don't think I have tried that myself. Uploading big files, yes but many of them...

What is the load average when the Pi is struggling ( from `top` or `uptime` ). Have you checked with `iotop`?

Windyo, 2017-08-11:
IOtop doesn’t exist, but top reports that CPU load is around 50% and mem load is around 5%.

nachoparker, 2017-08-11:
iotop you can install `sudo apt-get install iotop`

I am not talking CPU load only, but system load. Output of `uptime`?

Windyo, 2017-08-11:
up 56 min,  1 user,  load average: 8.26, 7.60, 7.77

nachoparker, 2017-08-11:
aha, load 8.26 is a LOT.

I think you are simply saturating the Pi. It is a non very powerful board to start with, plus you are in raid configuration writing to both USB ports at the same time.

That's why I suspected RAID1 was going to be a problem.

On top of that, ethernet also goes through the same PCI as the USB, so... recipe for desaster. You are using 3 ports ( 2 USB plus network ) on a single PCI lane.

Or are you on wifi?

It would be interesting to compare that to a single drive with no RAID

Windyo, 2017-08-11:
I should read up on load and such. Every time i dive into linux i realize I know Nothing.

Going through eth0, so you’re probably right on the « saturation » part.
I’ll leave it as-is for now, do some I/O tests, then break down the RAID and try again with AutoBackup.

And then if I have no other solution… Well I’ll just buy a small box to run nextcloud.
And probably realize I could Stream the box to my shield if I upgraded it a bit or Something.

nachoparker, 2017-08-11:
that's great, let us know

> And then if I have no other solution… Well I’ll just buy a small box to run nextcloud. And probably realize I could Stream the box to my shield if I upgraded it a bit or Something.

I understood nothing from that xDD

Windyo, 2017-08-11:
Basically if the PI isn't powerful enough to do what I expect it to, I'm going to buy a small pc to make it run.
Thing is with small PC's if you bulk em up enough they can become a no-display gaming platform which you can use to stream to something that can receive the signal - ShieldTV, SteamLink, whatever.

I'm basically saying that if I end up buying  asmall server for Nextcloud, it's going to end up as a nor-so-small gaming rig, streaming games ot my TV.... with nextcloud on a secondary drive, chugging along.

nachoparker, 2017-08-11:
ohh I see

We are discussing porting NextCloudPi to a more powerful board that is only $25 and looking good to us.

See https://help.https://help.nextcloud.com/t/best-cheap-hardware-to-run-nextcloud-on/4715/77nextcloud.com/t/best-cheap-hardware-to-run-nextcloud-on/4715/77

nachoparker, 2017-08-12:
Still interested to know if there is a difference in system load if you want to test it.

I'll close this now, but we can continue here

closed 2017-08-12 by nachoparker