nextcloud/nextcloudpi #107: Investigate RAID1 performance

nachoparker, 2017-08-11:
After starting this talk in #103, it would be interesting to try out a couple scenarios to see if it would be worth it.

 - `mdadm` RAID1
 - BTRFS RAID1
 - anything else

Thanks for the help @Windyo, anybody feel free to hop in

Windyo, 2017-08-11:
Soooooo... What exactly do I need to do ? Is there some kind of command-line tool I can run to test I/O on the array?

I mean we're going to need to test it with the same drives all the time, otherwise measurements make no sense, but other than that I have no idea how to test it.

nachoparker, 2017-08-11:
Well, I am not an expert, but I think you can try the "flexible I/O tester".

Also, you might want to look first if this study has already been done ;)

Windyo, 2017-08-14:
I've tested this a bit, and the results are as expected.

The bandwidth is split between the four USB ports and the ethernet controller, which makes adding a drive in Raid1 divide the total speed by... 2 in theory. In reality, I saw the performance cut down by 3.4 (average), probably due to saturation.

As-is, I can't really recommend the PI in a Raid 1 - It's viable as long as throughoutput is low (meaning you're not using it like I am). Connecting the Pi over Wi-fi would also solve part of the issue, but a cloud storage over WiFi is pushing it.

If I'm being honest this finding makes me think the Pi is an OK nextcloud testing ground, or for low-volume personal usage, but for any real-world use, the Pi is not going to be enough.

Windyo, 2017-08-14:
However, further testing shows that once data has been written to the disk, reading it and serving it is actually as fast (or a teeny tad faster) than without RAID1.

soooooooo.... Shitty to upload files, OK to query files. Still not worth it.

nachoparker, 2017-08-14:
Great!

Thanks very much for your testing

> If I'm being honest this finding makes me think the Pi is an OK nextcloud testing ground, or for low-volume personal usage, but for any real-world use, the Pi is not going to be enough.

Well, in my experience is more than fine for normal use. Of course that is my opinion. In any case, I am excited that now we are porting NextCloudPi to more powerful boards.

The rock64 has USB3 and  gigabit, for a comparable price ;)

Maybe we'll retest this once the porting is done

See

https://help.nextcloud.com/t/best-cheap-hardware-to-run-nextcloud-on/4715
http://uk.pcmag.com/news/90207/rock64-board-beats-raspberry-pi-on-price-and-performance

closed 2017-08-14 by nachoparker

MuellerSeb, 2020-10-20:
The Pi4 should do the trick, right?