nextcloud/nextcloudpi #108: Password is correct, authentication never works.

ObfuscateTheWeb, 2017-08-12:
### What version of NextCloudPi are you using? 
v0.19.11
### What is the base image that you installed on the SD card? 
NextCloudPi 12.0.1
### Expected behavior
Make changes to users (change password, change username, add user, etc) via admin account and put in the correct password to make changes. 
### Actual behaviour
After putting in correct password, I will be repeatedly asked to put in my password. This prompt keeps coming up even if my admin password is correct, authentication never works:

[image]

### Steps to reproduce, from a freshly installed image

1. Log in to admin account.
2. Go to users
3. Now try changing password, change username, etc.
4. The prompt for password will be repeated indefinitely even if password is correct.
### Include logs
<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"WY1YtH8AAQEAAAPp@Q8AAEUA","level":2,"time":"2017-08-11T07:11:48+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1Yu38AAQEAAAPp@RUAAEYG","level":2,"time":"2017-08-11T07:11:56+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1Yvn8AAQEAAAPp@RkAAEcK","level":2,"time":"2017-08-11T07:11:59+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1ZlH8AAQEAAAPoo9MAAA0D","level":2,"time":"2017-08-11T07:15:32+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1ZmH8AAQEAAAPoo9gAAAkF","level":2,"time":"2017-08-11T07:15:37+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1Zp38AAQEAAAPoo9wAABYO","level":2,"time":"2017-08-11T07:15:52+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1eZX8AAQEAAAPp@RwAAEkM","level":2,"time":"2017-08-11T07:36:06+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1ecX8AAQEAAAPopAAAABAM","level":2,"time":"2017-08-11T07:36:19+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1efH8AAQEAAAPp@SAAAEoQ","level":2,"time":"2017-08-11T07:36:32+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY1olX8AAQEAAAPopCYAABMD","level":2,"time":"2017-08-11T08:19:34+00:00","remoteAddr":"203.106.81.196","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'user1' (Remote IP: '203.106.81.196')","userAgent":"Mozilla\/5.0 (Android 7.1.1; Mobile; rv:55.0) Gecko\/55.0 Firefox\/55.0","version":"12.0.1.5"}
{"reqId":"WY1onH8AAQEAAAPopCgAAAMY","level":2,"time":"2017-08-11T08:19:41+00:00","remoteAddr":"203.106.81.196","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '203.106.81.196')","userAgent":"Mozilla\/5.0 (Android 7.1.1; Mobile; rv:55.0) Gecko\/55.0 Firefox\/55.0","version":"12.0.1.5"}
{"reqId":"WY1oon8AAQEAAAPopCoAAAIF","level":2,"time":"2017-08-11T08:19:47+00:00","remoteAddr":"203.106.81.196","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '203.106.81.196')","userAgent":"Mozilla\/5.0 (Android 7.1.1; Mobile; rv:55.0) Gecko\/55.0 Firefox\/55.0","version":"12.0.1.5"}
{"reqId":"WY11qH8AAQEAAAP1argAAEkY","level":2,"time":"2017-08-11T09:15:21+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY11r38AAQEAAAP1ar0AAEoG","level":2,"time":"2017-08-11T09:15:30+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY11t38AAQEAAAP1asEAAEsI","level":2,"time":"2017-08-11T09:15:39+00:00","remoteAddr":"192.168.1.1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY119H8AAQEAAAPzD4UAAAYG","level":2,"time":"2017-08-11T09:16:43+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY12CX8AAQEAAAPzD4kAAAcN","level":2,"time":"2017-08-11T09:17:10+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY12M38AAQEAAAPzD40AAAgP","level":2,"time":"2017-08-11T09:18:05+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=user1","message":"Login failed: 'user1' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY12yX8AAQEAAAPzD44AAAAJ","level":2,"time":"2017-08-11T09:20:10+00:00","remoteAddr":"60.191.38.77","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"60.191.38.77\" tried to access using \"175.141.42.33:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.11; rv:47.0) Gecko\/20100101 Firefox\/47.0","version":"12.0.1.5"}
{"reqId":"WY17LH8AAQEAAAP1auUAAFMM","level":3,"time":"2017-08-11T09:38:59+00:00","remoteAddr":"192.168.1.112","user":"user1","app":"PHP","method":"GET","url":"\/index.php\/apps\/files\/","message":"Invalid argument supplied for foreach() at \/var\/www\/nextcloud\/lib\/private\/Template\/SCSSCacher.php#145","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
{"reqId":"WY17j38AAQEAAAP1az4AAABC","level":2,"time":"2017-08-11T09:40:32+00:00","remoteAddr":"184.105.139.69","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.69\" tried to access using \"175.141.42.33\" as host.","userAgent":"--","version":"12.0.1.5"}
[... cut ...]
2017-08-11 22:14:43 1989570560 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-11 22:14:43 1989570560 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-11 22:14:43 1989570560 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

The weirdest thing is that if you were to do the same thing in mobile, it will work ! 
So far, I've been adding users and changing username/passwords via my android phone.

I suspect this is a bug..

nachoparker, 2017-08-12:
Duplicate of #11 

Make sure the  date on your RPi is correct.

closed 2017-08-12 by nachoparker

ObfuscateTheWeb, 2017-08-12:
I did raspi-config and select my timezone, I then run the date command and it returns the correct date and time. I reboot and same problem.

How did you do it?

ObfuscateTheWeb, 2017-08-12:
Tried manually setting date time with `date --set="12 August2017 2:28:00"`

Reboot and still doesn't work...

nachoparker, 2017-08-12:
have you tried without rebooting? RPi doesn't have an RTC

ObfuscateTheWeb, 2017-08-12:
Doesn't work either. I tried clearing the browser cache, tried it on chrome too and still doesn't work.

ObfuscateTheWeb, 2017-08-12:
What am I doing wrong? 
How exactly did you do it to make it work?

raspi-config and set timezone + reboot doesn't work
raspi-config and set timezone + no reboot doesn't work

manual set with `date --set="12 August 2017 2:28:00"` + reboot doesn't work
manual set with `date --set="12 August 2017 2:28:00"` + no reboot doesn't work

Is it as simple as changing time/date? Doesn't work on my end

ObfuscateTheWeb, 2017-08-13:
Okay, couldn't get it to work, but I've found a solution: Add users by using the occ command. It sucks this doesn't work on the GUI though, it seems that nextcloud is still in its infancy of development and it's still very buggy.

nachoparker, 2017-08-14:
Have you tried setting the date to some time in the future? The thing is that this is a security mechanism. 

When you input the password, NC will remember that you did for some minutes.

If your date was wrong at some point, NC will get confused. This should be improved, yes.

You might want to report it to nextcloud/server.

ObfuscateTheWeb, 2017-08-14:
How is this a security feature? I don't see how it could help with security..

Could you explain to me the reasoning behind it?

nachoparker, 2017-08-14:
Well, it's pretty standard ( for instance, when you remove a repo or an image in github or dockerhub).

In case you are victim of a CSRF attack, the attacker can make use of the fact that you are already authenticated in your browser with a session cookie, and perform evil.

For that reason, if you haven't input your password in a while, the system will ask you even if you are logged in, just to prevent that.

So the system needs to know when was the last time you input your admin password. If it happened under an incorrect date... well you can imagine the rest ^^

You can read more [here](https://ownyourbits.com/2017/03/25/nextcloud-a-security-analysis/)

ObfuscateTheWeb, 2017-08-15:
Link was an interesting read, it all makes sense now, thanks for clearing it up. 

Hey ! It works now?...

That's really odd that it didn't work before...
Oh well, glad it's finally working, but now I notice at my fail2ban isn't working anyone? 
Is it possible that fixing this issue also caused fail2ban to not work anymore as a side-effect?

ObfuscateTheWeb, 2017-08-15:
Oh and I noticed there's a new update of nextcloudpi to v0.20.2, where can I find the changelog?

nachoparker, 2017-08-15:
> where can I find the changelog?

Well, you can see all changes [here](https://github.com/nextcloud/nextcloudpi/commits/master)

Unfortunately the version does not show up there. You can see the versions [here](https://github.com/nextcloud/nextcloudpi/releases)

---

I just tested fail2ban from a clean image and it works. Open an issue if it doesn't work for you and we can investigate what's going on

nachoparker, 2017-09-05:
Duplicate of #11