nextcloud/nextcloudpi #109: symbolic link to usb drives not working reliably

candide10, 2017-08-13:
I have two usb drives connected (one for backup) and I just rebooted.  The symbolic links are now reversed.  The symbolic link "ncdrive" is pointing to my backup drive "ncbackup" and vice versa.  You should be able to see that here in this screenshot:
[image]

I have the auto mount function turned on.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.18.6

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
It's from about a month ago - that is mid to late July.

candide10, 2017-08-13:
The nextcloudpi image that I'm using is: 6/28/17

BTW - I rebooted a few times and still had the problem.  Then I manually deleted and then recreated the symbolic links and then rebooted and now things are working.  But doesn't the actual drive mapping supposedly switch around on successive reboots?

nachoparker, 2017-08-14:
Hi,

Thanks for reporting that. I am reviewing the automount feature. Initially it was just meant to be used with one drive, so that inexperienced linux users did not have to deal with fstab.

Now it's time to give it some more attention, as there are things to polish.

candide10, 2017-08-15:
OK.  Thanks.

nachoparker, 2017-08-15:
Ok, I fixed that issue along with other things in `v0.20.3`

Still, I am not very happy with the automount feature. Maybe we should go for something like `udiskie`

At this point, still only recommended for basic usage and unexperienced users.

 - Doesn't work well with a drive with several partitions
 - It duplicates directories under `/media/`

BTW, one of your drives (`sdb` adove) does not have a partition table. This is not recommended, even though often it works, but it is considered bad practice and I think it won't be detected after the changes I just made.

closed 2017-08-15 by nachoparker