nextcloud/nextcloudpi #110: How to expand storage space?

ObfuscateTheWeb, 2017-08-13:
Quick question, how do you expand your storage space when you run out?

Let's say my 1TB of storage runs out, how do I add more storage? Do you simply plug in another external USB drive? 

How does nextcloudpi handle this? From my understanding, each new usb is automounted on a separate directory, and the usb drive in question needs to be mounted on the datadir for it to work, how can you have two usb drives mounted on the same directory and have them both work together? Sounds like RAID...

Enlighten me  👍

nachoparker, 2017-08-14:
If you want to do that, you have to do it in a hacky way.

You can either use LVM to make both drives look like one bigger one, or you can mount the second drive in a directory inside the nextcloud data directory on the first drive, or maybe with a symbolic link.

None of those are super clean

closed 2017-08-14 by nachoparker

ObfuscateTheWeb, 2017-08-14:
Sounds like a future feature request maybe?

nachoparker, 2017-08-14:
Well, you can propose it but sounds quite out of scope... Most people would just get a bigger hard drive, or do the second mount manually

But yeah, propose it, we can start a discussion, or maybe somebody else comes up with a nice idea ;)

ObfuscateTheWeb, 2017-08-15:
Hmm..well if you think it's quite out of scope, then it's okay...

It would be nice though, but I also don't mind getting a bigger hard drive or doing the hacky method you mentioned.

If I actually run out of space, I think I'll get a 6 or 8TB drive haha...
Maybe put this in the bottom of your priority list? Maybe a very very very future release?

nachoparker, 2017-08-15:
haha sure, added to the whishlist