nextcloud/nextcloudpi #111: Stuck in mainataince mode

deljones, 2017-08-13:
Tried to do the upgrade that includes web admin for nextcloudpi

Flashed the image to a new SD card
Plugged in PI
set to mount the drive etc.
did a restore
Restore seemed to go ok
Now stuck on MAINTAINACE MODE
Cant get permission to the nextcloud DIR to turn mode off...

Thanks

Del

nachoparker, 2017-08-13:
Try

```
sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off
```

So, no errors restoring?

deljones, 2017-08-14:
Hello nachoparker

No error showed up.. I'll try it again later, I've gone back to the 
previous install for now.

This is what I did...

Used nc-backup to do backupflashed a blank SD card with the new 
imageInserted it into the pi and restarted.used nextcloudpi-config 
to:automount the USB HDD which has my data only, everything else is on 
the SD card, swap etc.nc-resotre to the last backupWaited, 'previous 
instance overwritten', files restore, database restore. Followed a long 
list of directories ended with 'press any key'Used browser to go to to 
my instance. Maintenance Mode splash screen. :(Tried a reboot. The same. 
:(Tried to get access to nextcloud DIR to turn off Maint Mode with occ, 
'permission denied.' :(
Will try sudo -u www-data php /var/www/nextcloud/occ maintenance:mode 
--off which I did try but I missed out -u, doh!

Everything seemed to go ok until the Maint mode.

I'll report back

Many many thanks

Del

<author@noreply.github.com>
Sent: 13-Aug-17 11:39:22 PM

deljones, 2017-08-14:
Hi Nachoparker….

 

Yep that did the trick…. Thanks again.

 

I have a question however, I’ve looked around for a solution on github about this.

 

I’d like the autobackup to go to another USB HDD. What would the safe process for this be?

 

Unplug my current data drive, plug new backup drive in, format it, name it, automount it. Plug back in existing drive, go to autobackup and send the back up to the newly named second drive?

 

I love this project by the way… Brilliant work.. Thank you

 

D

nachoparker, 2017-08-14:
@deljones 

Great, I am glad that it worked. 

I am a bit confused though with the fact that maintenance mode was on. That shouldn't happen.

 - Did you use `INCLUDEDATA` in your backup? In that case, when you are restoring, you should se a "restore datadir" message. Did you?
 - The "long list of directories"... I don't know what that is. If it ever happens to you again, could you paste it here?

> Unplug my current data drive, plug new backup drive in, format it, name it, automount it. Plug back in existing drive, go to autobackup and send the back up to the newly named second drive?

That is exactly right :)

I hope you like the web interface! Still working on it ;)

deljones, 2017-08-14:
nP

 

Yes I didn’t expect it to go into maint mode… so it was a bit of a surprise.

 

Yes I did do INCLUDEDATA and I did the a restore datadir as well…

 

I have my full system back with all my stuff, calendars and contacts on the new image. Missing two files that I downloaded from mails that missed out on the backup but they will go back in later.

 

Yes like the web UI. Very very useful… Great job dear chap!

 

D

nachoparker, 2017-08-14:
Ok, so for the rest it worked... I'll keep an eye for this. Please, anybody reopen if this happens again

closed 2017-08-14 by nachoparker