nextcloud/nextcloudpi #112: Letsencrypt .well-known acme-challenge timeout - How to fix?

nuke-it-newport, 2017-08-13:


closed 2017-08-14 by nachoparker

adriengibrat, 2018-02-10:
Same issue here, whith some logs, hopes it can help:

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.46.4
NextCloudPi image    NextCloudPi_11-22-17
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      1.3G/916G
rootfs usage         1.7G/3.7G
swapfile             /media/myCloudDrive/swap
Nextcloud check      ok
Nextcloud version    13.0.0.14
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.0.42
gateway              192.168.0.254
interface            enx001e0630cac8
certificates         none
certs due            none
NAT loopback         no
uptime               14min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.0.42",
            "6": "gigitipot.freeboxos.fr"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "13.0.0.14",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
[... 122 lines cut ...]
< Cache-Control: no-store, no-cache, must-revalidate
< Pragma: no-cache
< Content-Security-Policy: default-src 'self'; script-src 'self' 'unsafe-eval' 'nonce-XXXXXX'; style-src 'self' 'unsafe-inline'; frame-src *; img-src * data: blob:; font-src 'self' data:; media-src *; connect-src *; object-src 'none'; base-uri 'self';
< X-Frame-Options: SAMEORIGIN
< X-XSS-Protection: 1; mode=block
< X-Content-Type-Options: nosniff
< X-Robots-Tag: none
< X-Download-Options: noopen
< X-Permitted-Cross-Domain-Policies: none
< Set-Cookie: ocuh990ujuea=XXXXXX; path=/; secure; HttpOnly
< Set-Cookie: oc_sessionPassphrase=XXXXXX; path=/; secure; HttpOnly
< Set-Cookie: __Host-nc_sameSiteCookielax=true; path=/; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=lax
< Set-Cookie: __Host-nc_sameSiteCookiestrict=true; path=/; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=strict
< Strict-Transport-Security: max-age=15768000; includeSubDomains; preload
< Upgrade: h2,h2c
< Connection: Upgrade
< Location: https://gigitipot.freeboxos.fr/index.php/login
< Content-Length: 0
< Content-Type: text/html; charset=UTF-8
< 
* Curl_http_done: called premature == 0
* Connection #0 to host gigitipot.freeboxos.fr left intact

```
</details>

adriengibrat, 2018-02-10:
It seems it was domain provider issue (related to IPv6?), since I use a no-ip address, it works perfectly now...