nextcloud/nextcloudpi #113: add option re-active smb1-protocol?

JimmyKater, 2017-08-14:
as we talked about it in the comments on your homepage --> https://ownyourbits.com/2017/04/16/share-your-files-in-your-lan-with-samba/

would it be possible to add an option to ncpi to deactivate smb1 as a protocol using samba? or even better - for security purposes - to generally deactivate it by default and to add an option for users to re-activate it if their setting would demand using such an outdated security-hole?

nachoparker, 2017-08-14:
Definitely, it was on my list.

will do!

JimmyKater, 2017-08-14:
purrrrrrrrfect! :meouw:

nachoparker, 2017-08-29:
implemented in `v0.24.5`. I will have to update the blog, because the options to mount differ a little bit

Thanks!

closed 2017-08-29 by nachoparker

nachoparker, 2017-08-30:
Hi again,

I was testing this and I decided to roll it back.

 - There is no known vulnerability as far as I know with using SMB1
 - SMB2 and SMB3 don't support 'unix extensions', so it makes it more complicated to use

For instance, smbclient needs

```
smbclient -L 192.168.0.130 -U pi -mSMB2
```

and fstab

```
... -o vers=3.0,uid=1000,gid=1000
```

Which is not very clean

If anybody has other reason to disable SMB1, it can be done, just tell me what I am missing ;)

nachoparker, 2017-09-04:
implemented `v0.26.8`

Now it's all SMB3. Make sure to use the parameters above to mount

nachoparker, 2017-09-04:
actually I found a way to use SMB3 and not need those options

all good!