nextcloud/nextcloudpi #114: dyndns with own domain?

tommyguuuun, 2017-08-15:
Hi,
all is set up and ist working flawless again. My problem is now, that i want to update my dyndns domain (hostet at the german provider strato).
NextcloudPi only includes NoIp and when i try to install ddclient, it tells me that i need perl5, but perl is already installed. Maybe because a unstable version is used.

Is there any workaround for this?

JimmyKater, 2017-08-15:
hey tommy... 
just to jump in before @nachoparker is gonna answer - with a totally different thought.
which router do you run at home? fritzbox?
and do you have a domain of your own? 
if both are yes... set up a myfritz-account and set a CNAME entry in your domain. so that you could access your NC as _myCNAME.myDomain.myTLD_

nachoparker, 2017-08-15:
@tommyguuuun You can install `ddclient`, but you will have to do so from stretch.

Try 

```
sudo apt-get install -t stretch ddclient
```

@JimmyKater thanks for the info ;)

closed 2017-08-15 by nachoparker

JimmyKater, 2017-08-15:
@nachoparker it works. i am using this setup i mentioned above. pretty cool. and way better than any dyndns-service. imho

nachoparker, 2017-08-15:
yeah, that's pretty cool