nextcloud/nextcloudpi #115: Can't upload/create file is some Folder

Avelec, 2017-08-15:
Hi,

Do you know why some of the folders in nextcloud show message:
"You don't have permission to upload or create files here"

I have attached a picture of the screen shot. Thanks.
[image]

nachoparker, 2017-08-15:
You probably uploaded that folder externally? Through the terminal, SAMBA, NFS... ?

The folder (everything in your datadir, really) needs to be owned by `www-data` user

closed 2017-08-15 by nachoparker

Avelec, 2017-08-15:
Yes, I plugged the 2TB hdd to pc and copied all the folders/files. Then plugged it back to RP3. Some of the folders are ok but some are not (No permission to do anything with the files/folders). 

This is what I got:
pi@raspberrypi:~ $ ls -ld /media/USBdrive/ncdata/admin/files/
drwxrwxr-x 5 www-data www-data 4096 Aug 15 00:51 /media/USBdrive/ncdata/admin/fi                                                                                                                               les/

How can I make all the files/folders owned by www-data user? Thanks,

nachoparker, 2017-08-15:
Try
```
sudo chown -R www-data:www-data /media/USBdrive/ncdata
```

Avelec, 2017-08-16:
Thanks again for all your helps.

nachoparker, 2017-08-16:
;)