nextcloud/nextcloudpi #116: fail2ban does not work. Nothing changed on server side.

ObfuscateTheWeb, 2017-08-15:
### What version of NextCloudPi are you using?
v0.20.2
### What is the base image that you installed on the SD card? 
12.0.1
### Expected behavior
fail2ban behavior, normal behavior.
### Actual behaviour
fail2ban does not work but used to.
And the command `sudo fail2ban-client status nextcloud` does not show updated counters even if logged in with wrong username/password. Counters are always at 0, nothing changes, I tried running fail2ban again from the web address (deactivate and then reactivate), reboot, and still not working.
### Steps to reproduce, from a freshly installed image
I don't think this is reproducible, supposedly it just happens for no reason 
### Include logs
<details>
<summary>Nextcloud logs (Did not strip out my public IP, too lazy, too many lines)</summary>

```

Warning | core | Login failed: 'user1' (Remote IP: '192.168.1.112') | 2017-08-15T19:03:23+0800
Warning | core | Trusted domain error. "54.183.193.221" tried to access using "175.141.42.33" as host. | 2017-08-15T17:14:45+0800
Warning | core | Trusted domain error. "184.105.247.252" tried to access using "175.141.42.33" as host. | 2017-08-15T15:23:12+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.132.156.39') | 2017-08-15T11:17:37+0800
Warning | core | Trusted domain error. "169.229.3.91" tried to access using "175.141.42.33" as host. | 2017-08-15T10:48:58+0800
Warning | core | Login failed: 'user1' (Remote IP: '192.168.1.112') | 2017-08-15T08:56:10+0800
Warning | core | Login failed: 'admin' (Remote IP: '192.168.1.112') | 2017-08-15T08:52:53+0800
Warning | core | Login failed: 'admin' (Remote IP: '192.168.1.112') | 2017-08-15T08:50:49+0800
Warning | core | Login failed: 'admin' (Remote IP: '192.168.1.112') | 2017-08-15T08:49:34+0800
Warning | core | Login failed: 'admin' (Remote IP: '192.168.1.112') | 2017-08-15T08:25:27+0800
Warning | core | Login failed: 'admin' (Remote IP: '192.168.1.112') | 2017-08-15T08:25:18+0800
Warning | core | Trusted domain error. "60.191.38.77" tried to access using "175.141.42.33:443" as host. | 2017-08-15T07:19:16+0800
Warning | core | Trusted domain error. "184.105.247.196" tried to access using "175.141.42.33" as host. | 2017-08-14T17:04:09+0800
Warning | core | Trusted domain error. "169.229.3.91" tried to access using "175.141.42.33" as host. | 2017-08-14T15:14:17+0800
Warning | core | Trusted domain error. "159.203.240.99" tried to access using "175.141.42.33" as host. | 2017-08-14T12:19:46+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.164.178.198') | 2017-08-14T12:15:17+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.164.178.198') | 2017-08-14T12:13:18+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.164.178.198') | 2017-08-14T12:12:29+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.164.178.198') | 2017-08-14T12:12:06+0800
Warning | core | Login failed: 'user1' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:52+0800
Warning | core | Login failed: 'dd' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:38+0800
Warning | core | Login failed: 'dd' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:31+0800
Warning | core | Login failed: 'dd' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:25+0800
Warning | core | Login failed: 'dd' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:20+0800
Warning | core | Login failed: 'dd' (Remote IP: '115.164.178.198') | 2017-08-14T12:11:14+0800
Warning | core | Login failed: 'beenee' (Remote IP: '192.168.1.112') | 2017-08-14T10:16:03+0800
Error | internet_connection_check | GuzzleHttp\Exception\RequestException: cURL error 18: transfer closed with 25572 bytes remaining to read | 2017-08-14T10:14:43+0800
Warning | core | Login failed: 'sdfsd' (Remote IP: '192.168.1.1') | 2017-08-14T08:47:29+0800
Warning | core | Login failed: 'sdfsd' (Remote IP: '192.168.1.1') | 2017-08-14T08:46:12+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:43:36+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:43:06+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:42:35+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:42:05+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:41:35+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:41:05+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:39+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:26+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:20+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:16+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:15+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:40:12+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:37:08+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:37:02+0800
Warning | core | Login failed: 'djssj' (Remote IP: '192.168.1.1') | 2017-08-14T08:36:52+0800
Warning | core | Trusted domain error. "185.94.252.27" tried to access using "175.141.42.33" as host. | 2017-08-14T05:47:18+0800
Warning | core | Trusted domain error. "71.6.146.186" tried to access using "175.141.42.33" as host. | 2017-08-14T02:50:28+0800
Warning | core | Trusted domain error. "139.162.113.204" tried to access using "175.141.42.33" as host. | 2017-08-14T02:02:03+0800
Warning | core | Login failed: 'user1' (Remote IP: '192.168.1.112') | 2017-08-13T21:21:31+0800
Warning | core | Login failed: 'user12' (Remote IP: '192.168.1.156') | 2017-08-13T20:35:29+0800
Warning | core | Login failed: 'user12' (Remote IP: '192.168.1.156') | 2017-08-13T20:35:29+0800
Warning | core | Login failed: 'user12' (Remote IP: '192.168.1.156') | 2017-08-13T20:34:29+0800
Warning | core | Login failed: 'user12' (Remote IP: '192.168.1.156') | 2017-08-13T20:33:46+0800
Warning | core | Login failed: 'user1' (Remote IP: '192.168.1.156') | 2017-08-13T20:33:14+0800
[... 278 lines cut ...]
2017-08-15  8:51:03 1639437296 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-08-15  8:51:04 1639437296 [Note] InnoDB: Shutdown completed; log sequence number 30166948
2017-08-15  8:51:04 1639437296 [Note] /usr/sbin/mysqld: Shutdown complete

2017-08-15  8:51:20 1989193728 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-08-15  8:51:20 1989193728 [Note] InnoDB: The InnoDB memory heap is disabled
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-08-15  8:51:20 1989193728 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Using Linux native AIO
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Using generic crc32 instructions
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Completed initialization of buffer pool
2017-08-15  8:51:20 1989193728 [Note] InnoDB: Highest supported file format is Barracuda.
2017-08-15  8:51:21 1989193728 [Note] InnoDB: 128 rollback segment(s) are active.
2017-08-15  8:51:21 1989193728 [Note] InnoDB: Waiting for purge to start
2017-08-15  8:51:21 1989193728 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 30166948
2017-08-15  8:51:21 1447031792 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-08-15  8:51:21 1989193728 [Note] Plugin 'FEEDBACK' is disabled.
2017-08-15  8:51:22 1989193728 [Note] Server socket created on IP: '127.0.0.1'.
2017-08-15  8:51:22 1989193728 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

nachoparker, 2017-08-15:
I replaced the IPs for you.

Did you move the datadir?

ObfuscateTheWeb, 2017-08-15:
Oh yes, but that's the only thing major that I changed.

nachoparker, 2017-08-15:
can you do `tail -f /var/www/nextcloud/data/nextcloud.log` ( or wherever you configured logs when you set up fail2ban ) and then refresh the Nextcloud webpage, and tell me if you see new lines comming in as you do so?

ObfuscateTheWeb, 2017-08-15:
Yes, there are new lines coming in

ObfuscateTheWeb, 2017-08-15:
Updated to nextcloud 12.0.2 via the command line 
`cd /var/www/nextcloud/updater` then `sudo -u www-data php updater.phar`

nachoparker, 2017-08-15:
output of 

```
sudo fail2ban-client status nextcloud
```
?

nachoparker, 2017-08-15:
> Updated to nextcloud 12.0.2Updated to nextcloud 12.0.2

So, you are on 12.0.2 all this time?

ObfuscateTheWeb, 2017-08-15:
`sudo fail2ban-client status nextcloud`  no updates on counter.

No, I just updated as we speak

nachoparker, 2017-08-15:
:S that's going to make investigating this harder, as the log messages might have changed from version to version.

Still, what is the output of `sudo fail2ban-client status nextcloud`? I know it's 0 on the counters, but I want to see the log path

ObfuscateTheWeb, 2017-08-15:
Status for the jail: nextcloud
|- filter
|  |- File list:        /var/www/nextcloud/data/nextcloud.log
|  |- Currently failed: 0
|  `- Total failed:     0
`- action
   |- Currently banned: 0
   |  `- IP list:
   `- Total banned:     0

nachoparker, 2017-08-15:
thanks, can you paste for me one of the lines in `/var/www/nextcloud/data/nextcloud.log` that is like this one

```
{"reqId":"WZMHm38AAQEAAALVHyMAAAEC","level":2,"time":"2017-08-15T14:39:24+00:00","remoteAddr":"192.168.0.13","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'aoeu' (Remote IP: '192.168.0.13')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.1.5"}
```

ObfuscateTheWeb, 2017-08-15:
`{"reqId":"WZMHBn8AAQEAAAPTb5wAAA8W","level":2,"time":"2017-08-15T14:37:01+00:00","remoteAddr":"192.168.1.112","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'admin' (Remote IP: '192.168.1.112')","userAgent":"Mozilla\/5.0 (Windows NT 6.3; Win64; x64; rv:54.0) Gecko\/20100101 Firefox\/54.0","version":"12.0.2.0"}`

nachoparker, 2017-08-15:
It's weird, that should match. It is exactly the same ... btw it's better if you surround your code by triple `

Like this 

<pre>
 ```
code
 ```
</pre>

;)

ObfuscateTheWeb, 2017-08-15:
Okay, I'll remember that 👍 

> It's weird, that should match
Okay, both our outputs are the same..what now?

nachoparker, 2017-08-16:
weird, weird... let's try the fail2ban log

```
tail /var/log/fail2ban.log
```

ObfuscateTheWeb, 2017-08-16:
```

2017-08-16 20:23:07,928 fail2ban.actions[921]: INFO    Set banTime = 3600
2017-08-16 20:23:08,115 fail2ban.jail   [921]: INFO    Creating new jail 'nextcloud'
2017-08-16 20:23:08,115 fail2ban.jail   [921]: INFO    Jail 'nextcloud' uses pyinotify
2017-08-16 20:23:08,127 fail2ban.jail   [921]: INFO    Initiated 'pyinotify' backend
2017-08-16 20:23:08,137 fail2ban.filter [921]: INFO    Added logfile = /var/www/nextcloud/data/nextcloud.log
2017-08-16 20:23:08,140 fail2ban.filter [921]: INFO    Set maxRetry = 3
2017-08-16 20:23:08,142 fail2ban.filter [921]: INFO    Set findtime = 600
2017-08-16 20:23:08,145 fail2ban.actions[921]: INFO    Set banTime = 3600
2017-08-16 20:23:08,172 fail2ban.jail   [921]: INFO    Jail 'ssh' started
2017-08-16 20:23:08,178 fail2ban.jail   [921]: INFO    Jail 'nextcloud' started

```

nachoparker, 2017-08-16:
weird!!!

```
locale -a 
```
?

ObfuscateTheWeb, 2017-08-16:
```
C
C.UTF-8
en_GB.utf8
POSIX
```

New command..don't know it's output, what'd you think?

nachoparker, 2017-08-16:
lol everything seems ok. Plus it works on my end. Let's see this

```
fail2ban-regex /var/www/nextcloud/data/nextcloud.log /etc/fail2ban/filter.d/nextcloud.conf
```

ObfuscateTheWeb, 2017-08-16:
```

Running tests
=============

Use   failregex file : /etc/fail2ban/filter.d/nextcloud.conf
Use         log file : /var/www/nextcloud/data/nextcloud.log

Results
=======

Failregex: 113 total
|-  #) [# of hits] regular expression
|   1) [113] Login failed.*Remote IP.*'<HOST>'
`-

Ignoreregex: 0 total

Date template hits:
|- [# of hits] date format
|  [540] ISO 8601
`-

Lines: 540 lines, 0 ignored, 113 matched, 427 missed
Missed line(s): too many to print.  Use --print-all-missed to print all 427 lines
```

nachoparker, 2017-08-16:
it works, you have 113 matches!

ObfuscateTheWeb, 2017-08-16:
So what if it has 113 matches? Nothing is being blocked, counters are not updating, it doesn't do its function

ObfuscateTheWeb, 2017-08-16:
well so far not much change from the default..

nachoparker, 2017-08-16:
Just to make sure, have you tried

```
sudo fail2ban-client status nextcloud
```
right after faking an incorrect login attempt?

ObfuscateTheWeb, 2017-08-16:
Yes, I always do to make sure my output is correct

nachoparker, 2017-08-16:
have you done any changes in your timezone configuration?

ObfuscateTheWeb, 2017-08-16:
Yes, I ran `sudo raspi-config` and changed my timezone to where I am.

I did this for the admin page to work, otherwise I'll have to use the occ command for admin stuff

nachoparker, 2017-08-16:
can you try
```
sudo service rsyslog restart
sudo service fail2ban stop
sudo service fail2ban start
```

ObfuscateTheWeb, 2017-08-16:
done..

nachoparker, 2017-08-16:
does it work now?

ObfuscateTheWeb, 2017-08-16:
no

nachoparker, 2017-08-16:
still nothing on `sudo fail2ban-client status nextcloud`?

Then I'll have to test myself what happens when you change the tzdata.

It looks really similar to [this](https://stackoverflow.com/questions/20720129/fail2ban-not-banning-ssh-bruteforce-but-regex-works) problem

ObfuscateTheWeb, 2017-08-16:
Yeah, it is really similar to that other issue, so it's timezone issues?

And yeah, nothing on `sudo fail2ban-client status nextcloud`

nachoparker, 2017-08-16:
Could be... I'll try again to reproduce with that info

Thanks for the feedback

ObfuscateTheWeb, 2017-08-16:
thanks too, you've been really responsive and helpful !

Yeah, try it and see, I'm curious too...

nachoparker, 2017-08-21:
I'll add the 'help wanted' tag in case somebody want to try to reproduce it playing with the timezone. Otherwise, I'll do it when I have time, but I am a little bit busy now

ObfuscateTheWeb, 2017-08-23:
I wonder, if I change the timezone back to the default settings, will fail2ban work again?
I mean, it did stop working after changes to the timezone right? Problem is, I don't know what was the default timezone setting.

And ``dpkg-reconfigure tzdata`` isn't particularly helpful either..
Do you know how to reset everything timezone-related back to defaults?

I don't mind if the web interface doesn't work, I'll just use the occ commands

nachoparker, 2017-08-23:
Well, the one that comes by default is

```
root@raspberrypi:/home/pi# cat /etc/timezone 
Etc/UTC
root@raspberrypi:/home/pi# timedatectl
      Local time: Wed 2017-08-23 15:44:31 UTC
  Universal time: Wed 2017-08-23 15:44:31 UTC
        RTC time: n/a
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
```
, so I guess UTC+0

I still haven't verified that TZ is the issue with fail2ban, I am sorry. But maybe you can try ;)

ObfuscateTheWeb, 2017-08-23:
Oh my god ! It works !

I did `sudo raspi-config` and select UTC. 
Reboot, run `sudo nextcloudpi-config` and run fail2ban again, log in with wrong credentials and did `sudo fail2ban-client status nextcloud`

And sure enough, the outputs are updated !

[image]

nachoparker, 2017-08-23:
Awesome! Good job!

I suspect that it started logging on a TZ, then after you switched back to a different TZ because NC was confused with the time, fail2ban became confused with the time.

I will try to reproduce and see if we can prevent this from happening but for now it will sit on low priority because it is a weird corner case ;)

nachoparker, 2017-08-30:
I'll close this for now, if anybody is able to reproduce, please reopen

closed 2017-08-30 by nachoparker