nextcloud/nextcloudpi #117: ND-DATADIR breaks NCP

Shadowstreik, 2017-08-15:
Issue Reference... #93 
[image]

In reference to the linked question, I performed NC_DATADIR to move all data from the SDCard to the external hard drive.

- Performed two backups, one without data and one with. A no-data backup takes 5 to 15 minutes, A data-included backup takes 15 to 18 hours (35-37 gigs).
- Kicked off NC-DATADIR to my USB to allow the amount of time a data-included backup.

Allowed a full 24 hours for process to complete. Checked and was notified it was in Mainteannce Mode. I shut off Maintenance Mode and ended up with the attached picture.

This is only a heads-up as I must now rebuild it and change the datadir after the no-data restore for the quickest resolution.

nachoparker, 2017-08-16:
Wow, you are really pushing NextCloudPi! 

That's great feedback

Did you run `nc-datadir` from `nextcloudpi-config`?

Did you capture the output from it? maybe the process was not finished

Shadowstreik, 2017-08-16:
I had a feeling that pushing a process for 35ish gigs might be a bit of a large expectation. Escpecially after running into the problem.

I used the web app to start the NC-DATADIR change process. With running a large backup through the web app, it would run about 5 minutes and then tell me something went wrong and tell me to possibly refresh the page. The process would still be running because I'd be connected with Filezilla and refresh the directory where the large backup was running and the size would be increasing. So I would let it continue to run.

What I ended up doing was my full rebuild using the 7/29 release, running updates for both NCP and then Nextcloud. I tried using a restore of the non-data backup, but it wouldn't show any folders. I did it over and decided to not include any video in the rebuild. That, alone, kept the total space use low. I also decided to forego the USB altogether and stay on, only, the SD card. It works best for my needs, there.

It, certainly, possible that the process wasn't finished, but it doesn't seem logical as a backup including data takes me 15 to 18 hours and it was still possibly not finished after 24. I wasn't worried and could take care of it from a different approach.

nachoparker, 2017-08-17:
Hi,

First, I recommend that you always restore (or do anything really) from the latest image, because there are little improvements that might not be covered by `ncp-update`
It surely would take a LONG time to achieve what you are trying to do. I will test with really big backups when I have time.
Also, if you restore a non-data backup, the files wont be there, and your datadir will be back at `/var/www/nextcloud/data`. You would have to move your datadir again and manually put the files back again.

Seems like the process was not over, and that's why you get that screen. 

Anyway, that's good info to continue improving. I'll test this when I have time. Also looking forward to NC13 when files over 2GB can be uploaded.

closed 2017-08-17 by nachoparker