nextcloud/nextcloudpi #118: Error setting up letsencrypt

fab1ton, 2017-08-15:
Referring to issue #81. I have the exact same issue the only difference is, that I already opened port 80 and 443. I am using ddclient with namecheap.
I actually just reinstalled NextcloudPi. Before everything was working just fine, just now I am facing this issue. Issue:
The following errors were reported by the server:

Domain: XXXXX Type: connection Detail: Fetching XXXXXXX/.well-known/acme-challenge/kcyhqhh4O7yZ_VbuXHCE05eszXBkXezL6e3lcQG9M58: Timeout

To fix these errors, please make sure that your domain name was entered correctly and the DNS A/AAAA record(s) for that domain contain(s) the right IP address. Additionally, please check that your computer has a publicly routable IP address and that no firewalls are preventing the server from communicating with the client. If you're using the webroot plugin, you should also verify that you are serving files from the webroot path you provided.

Your account credentials have been saved in your Certbot configuration directory at /etc/letsencrypt. You should make a secure backup of this folder now. This configuration directory will also contain certificates and private keys obtained by Certbot so making regular backups of this folder is ideal.

nachoparker, 2017-08-16:
Well, you have a timeout which hints some kind of connectivity problem. Can you access your domain from outside?

fab1ton, 2017-08-16:
I am able to reach my routers IP/ own domain with my browser. However if I ping my ip/domain I am getting a timeout.

I just figured out that it could be port 80. Was port 80 always needed in the previous NextcloudPi images? Because before I didn't even open port 80 and it worked.

I already tried to set up letsencrypt with port 443 only but unfortunately I wasn't able to do it. I checked other issue posts here which didn't help either.

nachoparker, 2017-08-16:
Right now you need both 80 and 443 ports open. Please, try that.

I think both were needed all the time, but I made some changes in the past, so I am not sure

closed 2017-08-16 by nachoparker

fab1ton, 2017-08-16:
So at this point right now I am not able to set up letsencrypt without port 80?

Do you maybe have somewhere the old image, which you can send me, so that I can run letsencrypt and afterwards update the system myself?

nachoparker, 2017-08-16:
Mmmm, issue this command and try again from `nextcloudpi-config`

```
wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/beb9bc1ee2909a1ab6bfde7398ddf19a50d02478/etc/nextcloudpi-config.d/letsencrypt.sh -O /usr/local/etc/nextcloudpi-config.d/letsencrypt.sh
```

fab1ton, 2017-08-16:
I will try that and keep you updated. Hopefully it will work *crossed fingers*

fab1ton, 2017-08-17:
It actually worked. Thanks a lot! You can mark this thread as solved and I can confirm, that you can successfully sign lets encrypt with only port 443.

nachoparker, 2017-08-17:
awesome ;)

DoTheEvo, 2018-01-11:
nachoparker answer works, commenting so I can find it next time

DoTheEvo, 2018-01-13:
`Client with the currently selected authenticator does not support any combination of challenges that will satisfy the CA`

lets encrypt [disabled](https://community.letsencrypt.org/t/solution-client-with-the-currently-selected-authenticator-does-not-support-any-combination-of-challenges-that-will-satisfy-the-ca/49983) TLS-SNI-01 challenge 324 due to a security report.

So workaround is not working for now it seems. Lets hope they solve it fast.