nextcloud/nextcloudpi #119: “shutdown” or “standby” function in the web interface

user-jan, 2017-08-16:
Hi Nacho,
Up to now I have always used ssh to turn off my NextCloudPi server during night time.
Would it be possible to include a “shutdown” or “standby” function in the web interface, or is this too complicated to do?

nachoparker, 2017-08-16:
Hi again Jan!

Sure, that sounds like a must-do! I had been thinking about that and also about turning on SSH from the web, once there is password authentication in place.

elessarelfstar, 2017-08-17:
Yes -- both features would be fantastic. Thanks for your work.

nachoparker, 2017-08-17:
Ok, seems like people really want this, so I prioritized ;)

First version, would probably be nice to have a confirmation dialog... careful for now!

[image]

Done

closed 2017-08-17 by nachoparker

user-jan, 2017-08-18:
Brilliant, thanks alot Nacho!!!

EtienneHenger, 2019-03-16:
hi, new here. so when I try and shutdown the ncp via web I receive the confirmation dialog that ncp is "Shutting down..." however, the status does not change and I am able to access the pi via ssh and see that ncp is still up and running on  the pi. A simple reload of the ncp-web brings the panel back with all the settings. Is there supposed to be a confirmation about the shutdown? I have let it sit for a couple hours without any effect (pi still being very active with less flashing). Also is there a way to shutdown the pi and not just the ncp service? That would be wonderful.

nachoparker, 2019-03-16:
shutdown means "shut down the pi", not the ncp service. It should shut down right away, I just tested it and it works

EtienneHenger, 2019-03-17:
thanks for the quick reply @nachoparker 👍 
It turned out that the pi was not responding to anything... no ssh, no ncp-web commands, and not even direct input via keyboard... so I had to force reboot it with a power cycle. This made me run into a new problem which I posted  here: 
    https://help.nextcloud.com/t/ncp-error-no-data-dir-found-after-reboot-of-pi/49675?u=pinewb
I am seeing why the pi is not the best option. 👎