nextcloud/nextcloudpi #122: Web GUI not fully populating

deljones, 2017-08-17:
Hi

Just signed in locally to the Web GUI only to find it empty :(

Anyone else seen this? 

Del
[image]

nachoparker, 2017-08-17:
the _responsiveness_ is not complete, meaning that it won't show the sidebar on a narrow browser, such as a phone... yet

deljones, 2017-08-17:
Understood 

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512

elessarelfstar, 2017-08-17:
If you are using an mobile browser, enable the desktop view in the settings menu. Everything will be small, but it will populate correctly.

nachoparker, 2017-08-17:
Thanks @elessarelfstar 

This is amongst the needed improvements to the web UI. Creating a web interface is quite time consuming... actually if anybody wanted to contribute with code this could be a good start

ObfuscateTheWeb, 2017-08-23:
@nachoparker why didn't you use bootstrap to develop it if responsive design was needed?

Wouldn't it have been much much easier?

nachoparker, 2017-08-23:
Hi @ObfuscateTheWeb 

I thought it would be an overkill to use bootstrap, vue or angular for such a simple interface

I am very much in favour for minimalism. I don't like the current trend of 2MB of javascript code for a hello world website, and web development is pretty messy nowadays.

That's why I chose minified.js instead of jquery. Of course it's a bit more work, but doesn't it load lightning fast ;)

The whole thing loads in ~100ms, being only 30KB. I wrote about it here

https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/

The code is in the `ncp-web` folder

https://github.com/nextcloud/nextcloudpi/blob/master/ncp-web/index.php

But that's just my personal preference. If somebody would be willing to help improve ncp-web another approach could be used ;)

In any case, for the issue at hand, it is just a matter of implementing the left sidebar, in the fashion the nextcloud web interface does, which is showing a button to hide/show the sidebar whenever the screen is too small (through media query I suppose)

deljones, 2017-08-23:
Yes I like the simple quick approach..:)

 

d

 

https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/

https://github.com/nextcloud/nextcloudpi/blob/master/ncp-web/index.php

ObfuscateTheWeb, 2017-08-23:
Well, I'm not an expert coder, and I see that you don't like bloated websites. Nevertheless, I've went on with my attempt at making a responsive web using bootstrap, definitely not the cleanest or best implementation, but it does the function of responsiveness. It's not complete with working buttons and the poweroff button is not present due to my lack of coding expertise, so you'll have to chime it and add the button functions, php, javascripts, etc...

Here's it is:

[image]

[image]

[image]

It's not much, so yep...  ¯\_(ツ)_/¯

nachoparker, 2017-08-23:
Boy!!! awesome!!! thank you very much!!!

I'll be a bit busy this week, but we can talk about that later. Thanks for your time!!

Did you do it on top of my code, or did you take a bootstrap template and plugged my code in, or... ?

ObfuscateTheWeb, 2017-08-23:
> Did you do it on top of my code, or did you take a bootstrap template and plugged my code in, or... ?

Err...don't know, but here's the source:

```
<!DOCTYPE html>
<html lang="en">
<head>
  <title>NextCloudPi Panel</title>
  <link rel="icon" type="image/png" href="favicon.png" />
  <meta charset="utf-8">
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="bootstrap.min.css">
  <link rel="stylesheet" href="sideNavbar.css">
  <script src="jquery.min.js"></script>
  <script src="bootstrap.min.js"></script>
</head>
<body>
<noscript>
  <div id="nojavascript"> <div>This application requires JavaScript for correct operation. Please <a href="http://enable-javascript.com/" target="_blank" rel="noreferrer">enable JavaScript</a> and reload the page.		</div> </div>
  </noscript>

<nav class="navbar navbar-inverse navbar-fixed-side">
  <div class="container-fluid">
    <div class="navbar-header">
      <button type="button" class="navbar-toggle" data-toggle="collapse" data-target="#myNavbar">
        <span class="icon-bar"></span>
        <span class="icon-bar"></span>
        <span class="icon-bar"></span>                        
      </button>
	   <a href="https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/" target=_blank><img class="navbar-brand img-responsive" src="ncp-logo.png"></a>
	   <br /><br /><strong><small class="text-muted">V0.24.1</small></strong>
	  
    </div>
   <div class="collapse navbar-collapse" id="myNavbar">
      <ul class="nav navbar-nav">
        <li class="dropdown">
          <a class="dropdown-toggle" data-toggle="dropdown"></a>
                                                                <li id="NFS"  ><a href="#"> NFS </a><input type="hidden" value="NFS network file system server (for Linux LAN)" /></li>
						  <li id="dnsmasq"  ><a href="#"> dnsmasq </a><input type="hidden" value="DNS server with cache" /></li>
						  <li id="fail2ban"  ><a href="#"> fail2ban  </a><input type="hidden" value="Brute force protection for SSH and NextCloud" /></li>
						  <li id="letsencrypt"  ><a href="#"> letsencrypt </a><input type="hidden" value="Automatic signed SSL certificates" /></li>
						  <li id="modsecurity"  ><a href="#"> modsecurity </a><input type="hidden" value="Web Application Firewall for extra security (experimental)" /></li>
						  <li id="nc-automount"  ><a href="#"> nc-automount  </a><input type="hidden" value="Automount USB drives by plugging them in" /></li>
						  <li id="nc-autoupdate-ncp"  ><a href="#"> nc-autoupdate-ncp  </a><input type="hidden" value="Automatically apply NextCloudPi updates" /></li>
						  <li id="nc-backup-auto"  ><a href="#"> nc-backup-auto </a><input type="hidden" value="Periodic backups" /></li>
						  <li id="nc-backup"  ><a href="#"> nc-backup </a><input type="hidden" value="Backup this NC instance to a file" /></li>
						  <li id="nc-database"  ><a href="#"> nc-database </a><input type="hidden" value="Move your database to a new location, like a USB drive" /></li>
						  <li id="nc-datadir"  ><a href="#"> nc-datadir </a><input type="hidden" value="Change your data dir to a new location, like a USB drive" /></li>
						  <li id="nc-format-USB"  ><a href="#"> nc-format-USB </a><input type="hidden" value="Format an external USB drive as a ext4 partition (dangerous)" /></li>
						  <li id="nc-forward-ports"  ><a href="#"> nc-forward-ports </a><input type="hidden" value="Set port forwarding to access from outside (UPnP)" /></li>
						  <li id="nc-httpsonly"  ><a href="#"> nc-httpsonly  </a><input type="hidden" value="Force HTTPS" /></li>
						  <li id="nc-init"  ><a href="#"> nc-init </a><input type="hidden" value="(Re)initiate Nextcloud to a clean configuration" /></li>
						  <li id="nc-limits"  ><a href="#"> nc-limits </a><input type="hidden" value="Configure system limits for NextCloudPi" /></li>
						  <li id="nc-news"  ><a href="#"> nc-news </a><input type="hidden" value="Install the latest News third party app" /></li>
						  <li id="nc-nextcloud"  ><a href="#"> nc-nextcloud </a><input type="hidden" value="Install any NextCloud version" /></li>
						  <li id="nc-notify-updates"  ><a href="#"> nc-notify-updates  </a><input type="hidden" value="Notify in NC when a NextCloudPi update is available" /></li>
						  <li id="nc-ramlogs"  ><a href="#"> nc-ramlogs </a><input type="hidden" value="mount logs in RAM to prevent SD degradation (faster, consumes more RAM)" /></li>
						  <li id="nc-restore"  ><a href="#"> nc-restore </a><input type="hidden" value="Restore a previously backuped NC instance" /></li>
						  <li id="nc-scan-auto"  ><a href="#"> nc-scan-auto </a><input type="hidden" value="Periodically scan NC for externally modified files" /></li>
						  <li id="nc-scan"  ><a href="#"> nc-scan </a><input type="hidden" value="Scan NC for externally modified files" /></li>
						  <li id="nc-swapfile"  ><a href="#"> nc-swapfile </a><input type="hidden" value="Move and resize your swapfile. Recommended to move to a permanent USB drive" /></li>
						  <li id="nc-update"  ><a href="#"> nc-update </a><input type="hidden" value="Update NextCloudPi" /></li>
						  <li id="no-ip"  ><a href="#"> no-ip </a><input type="hidden" value="Free Dynamic DNS provider (need account)" /></li>
						  <li id="samba"  ><a href="#"> samba </a><input type="hidden" value="SMB/CIFS file server (for Mac/Linux/Windows)" /></li>
						  <li id="unattended-upgrades"  ><a href="#"> unattended-upgrades  </a><input type="hidden" value="Automatic installation of security updates. Keep your cloud safe" /></li></ul><br />

          
        </li>
        
      </ul>
       
     
      <!--<ul class="nav navbar-nav navbar-right">
        <li><a href="#">Login</a></li>
      </ul> -->
    </div>
  </div>
</nav>
  
<div class="container text-center">
  <h4>Configure NextCloudPi features</h4>
</div>

</body>
</html>
```

nachoparker, 2017-08-23:
Awesome, thanks. I'll get back to you when I have time for this.

If you are up to, maybe you can push the code yourself and appear as a contributor ;)

ObfuscateTheWeb, 2017-08-23:
Welcome there, nacho

Push the code? Do I simply add the file to ncp-web on github?

nachoparker, 2017-08-23:
yes, I have to prepare that for you first, and we should talk a little bit about it before you do the pull request (PR). 

I'll explain to you how to do it ;)

I am now preparing to go to Berlin to the Nextcloud conference, so I don't have much time right now. Sorry!

We'll talk about this like in a week! ;)

ObfuscateTheWeb, 2017-08-23:
Awesome ! Have fun there !

nachoparker, 2017-08-23:
Thanks! I am excited ^^

nachoparker, 2018-02-19:
implemented, `v0.46.23`

closed 2018-02-19 by nachoparker