nextcloud/nextcloudpi #126: External HDD

matzel82, 2017-08-19:
I hope I understand, because I have to use the Google translator. I have already set up the Nextcloud, but unfortunately something went wrong and I had to do something new. Unfortunately, it is currently not possible to integrate an external hard drive into the system. Auto mount is enabled. And my old data is still on the hard drive. How do I proceed to integrate this disk into the system?

matzel82, 2017-08-19:
I have used the latest version.
The HDD is formated now ext4
Mounted as sda1 (lsblk)
Name is USBdrive1 an config in nc tool ncdata dir

nachoparker, 2017-08-20:
hi,

where are you from?

If you formated the drive... you have lost any data that was there.

If you only have one drive plugged in, you can use `/media/USBdrive/ncdata` in `nc-datadir` and your files will be stored there

closed 2017-08-20 by nachoparker

matzel82, 2017-08-20:
From Germany

The Drive is formatted, automatic mount.
The DiscSpace is now correct.

The data i have copy manual is on The cloud 
Too...

It works 

Sorry for my english

matzel82, 2017-08-27:
I Need please THW correct config for fstab.

I have only plug in 1 Drive (sda1)

After a reboot the mountpoint (USBdrive1...) is changed.

How do it ?