nextcloud/nextcloudpi #127: No web access after resizing partition

niconoob, 2017-08-19:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.23, just upgraded from version in image 06-28-17
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
image 2017-06-28
### Expected behavior
I had the error message on bash: "-bash: cannot create temp file for here-document: No space left on device"
I googled it up and expanded the file system using raspi-config. I expected I could access nextcloud via web to set it up.
### Actual behaviour
After resizing the file system, I cannot access mi pi via web, as described by a user in a comment under https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/
### Steps to reproduce, from a freshly installed image
Installed image, set up wi-fi, tried to set up letsencrypt (didn't work, dyndns not yet working apparently), checked my pi on the browser and saw the setup page from nextcloud, updated nextcloudpi, expanded the filesystem, came back to the web interface, it didn't work.

There are no useful nextcloud logs as I didn't even set it up.

Paste the output of `systemctl status apache2`
```apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled)
   Active: failed (Result: exit-code) since Sat 2017-08-19 21:23:14 UTC; 19min ago
  Process: 3276 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
```
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Sat Aug 19 06:25:03.000983 2017] [mpm_event:notice] [pid 728:tid 1991954432] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Sat Aug 19 06:25:03.001172 2017] [core:notice] [pid 728:tid 1991954432] AH00094: Command line: '/usr/sbin/apache2'
[Sat Aug 19 07:17:09.006053 2017] [mpm_event:notice] [pid 715:tid 1992876032] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Sat Aug 19 07:17:09.025241 2017] [core:notice] [pid 715:tid 1992876032] AH00094: Command line: '/usr/sbin/apache2'
[Sat Aug 19 20:36:42.216931 2017] [log_config:warn] [pid 735:tid 1740633136] (28)No space left on device: [client 192.168.178.26:34328] AH00646: Error writing to /var/log/apache2/other_vhosts_access.log
[Sat Aug 19 20:37:55.549429 2017] [mpm_event:notice] [pid 715:tid 1992876032] AH00491: caught SIGTERM, shutting down

==> /var/log/apache2/other_vhosts_access.log <==

```
Paste the output of `systemctl status mysqld`
```mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled)
  Drop-In: /run/systemd/generator/mysql.service.d
           └─50-mariadb-$database.conf
   Active: failed (Result: exit-code) since Sat 2017-08-19 21:12:12 UTC; 28min ago
 Main PID: 711 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"
```
Paste the output of `tail -n 100 /var/log/mysql/*.log`
``` [it is empty]```

nachoparker, 2017-08-20:
Hi, first look at #88 might be related

Second... let's check that you indeed have space left, because the logs indicate otherwise

 output of `df -h`?

niconoob, 2017-08-20:
issue #88 describes the actual problem, thanks!
I forgot to mention that I, too, had tried to set up lentsencrypt to no avail, so correcting those two lines in nextcloud.conf and restarting apache2 did the trick :)

output of `df -h` seems fine btw, that part of the logs must have been right before I resized. 

Thanks a lot!

closed 2017-08-20 by niconoob

nachoparker, 2017-09-05:
Duplicate of #88