nextcloud/nextcloudpi #128: Samba is not uploading to my cloud

fab1ton, 2017-08-20:
I am able to connect to samba, see all files but I am not able to delete/drag and drop/ move files. Furthermore if I am creating a new folder it is not updating into the cloud. In the web interface/ nextcloud client the changes are not coming up.

nachoparker, 2017-08-20:
how did you setup samba?

have you read this?

https://ownyourbits.com/2017/04/16/share-your-files-in-your-lan-with-samba/
https://ownyourbits.com/2017/04/18/different-ways-to-access-your-nextcloud-files/

Rufio87, 2017-08-23:
Hi fab1ton

I had the same issue. Could you please move to the folder, where your Samba-folder is stored do the following and post the output here (without the "): 
1. "ls -l"
2. "group username" - username is the name of the user you want to allow access to the samba-folder

fab1ton, 2017-08-24:
Hi,
The main problem I am facing now is that if I create a folder in Samba I am don't have write permission in Nextcloud. Same problem the other way around if I create a folder in Nextcloud I don't have write permission in Samba. 

Here the outtake: 
drwxr-xr-x  2 pi       pi       4096 Aug 20 04:18 FOLDER XXX
drwxr-xr-x  6 pi       pi       4096 May 26 01:43 FOLDER XXX
drwxr-xr-x  7 pi       pi       4096 Feb  9  2017 FOLDER XXX
drwxr-xr-x  2 www-data www-data 4096 Aug 22 21:38 FOLDER XXX
drwxr-xr-x 23 pi       pi       4096 Aug 21 00:17 FOLDER XXX
drwxr-xr-x  2 www-data www-data 4096 Aug 22 21:59 FOLDER XXX
drwxr-xr-x  8 pi       pi       4096 Aug 21 00:19 FOLDER XXX
drwxr-xr-x  2 pi       pi       4096 Mar 28 19:16 FOLDER XXX
drwxr-xr-x  6 pi       pi       4096 Aug 21 04:42 FOLDER XXX
drwxr-xr-x  3 pi       pi       4096 Aug 21 20:48 FOLDER XXX

Note that I have created most folders in Samba which means I dont have write permission in Nextcloud.

fab1ton, 2017-08-24:
Line 5 and 6 are created in Nextcloud, rest in Samba:
Line 5: drwxr-xr-x 2 www-data www-data 4096 Aug 22 21:38 FOLDER XXX
Line 6: drwxr-xr-x 2 www-data www-data 4096 Aug 22 21:59 FOLDER XXX

Rufio87, 2017-08-24:
first of all you need to activate nc-scan-auto in the nextcloud configuration (type "sudo nextcloudpi-config). In order to do that manually you can also select nc-scan. with this you will find all your changed data with smaba in nextcloud.

secondly, your samba user is pi and your nextcloud user is www-data. In your current setting you will only have restricted permissions to access in samba with your nextcloud client. probably you will only have read and no write permission. 

what exactly do you want? do you want full access of you samba files with your nextcloud client? If your answer is yes you could do the following: 

1.  change your samba config file by: "sudo nano /etc/samba/smb.conf

2.  change add/change following lines where you define your NAS folder(s):

[NAS_folder]
path = /here/is/the/path/of/your/NAS/folder
writeable = yes
browseable = yes
valid users = pi
force group = www-data 
create mask = 0770
directory mask = 0771

3. for all files and folders you already created with samba you need to set the owner group to www-data by "sudo chown :www-data name_of_file_or_folder

4. for all files and folders you already created with samba you need to change the owner group access rights by "sudo chmod g=rwx name_of_file_or_folder"

Now all newly added files with samba will be owned by the group www-data and the users of this group will have read, write and execute rightsy -> you will have also write access in nextcloud for all files and folders.

fab1ton, 2017-08-24:
I did what you did, added those lines to my smb.conf. Used the following command sudo chown :www-data /media/hdd/ncdata/admin/files/Documents but now my nextcloud is in maintenance mode and not available anymore. Attached a pic of my smb.conf

fab1ton, 2017-08-24:
[image]

fab1ton, 2017-08-24:
I actually could fix the maintenance mode and I am now able to read and write files which I created on Nextcloud with the following command you mentioned: sudo chmod g=rwx name_of_file_or_folder

I am just still not able to access files on Nextcloud created in Samba. I tried this command but it didnt help: sudo chown :www-data name_of_file_or_folder

Rufio87, 2017-08-24:
It went into maintenance mode by itself? This set up worked for me without issue. Try to restart your samba server and disable maintenance mode. 

It is important to understand the access rights of pi and www-data and that your NAS with samba works with pi and nextcloud works with www-data. The user www-data which is in the self named group www-data needs write access to all files created with samba.

what do you mean by accessing files on Nextcloud? Are they not visible in Nextcloud or do you have restricted permissions?

fab1ton, 2017-08-24:
Ok, for my understanding - this command "sudo chown :www-data name_of_file_or_folder" gives in Samba created folders the right to use them in Nextcloud, right?
And this command "sudo chmod g=rwx name_of_file_or_folder" gives folders created in Nextcloud the rights in Samba? I hope I am explaining myself good enough, its pretty confusing for me.

I mean that the files I created in Samba are not write-able in Nextcloud. I can see them I just cant create / modify files out of Nextcloud, just out of Samba. I already tried the following command: "sudo chown :www-data name_of_file_or_folder" This is the exact message I am getting: "You dont have permission to upload or create files here"

-Your help is much appreciated, thanks a lot for helping me!

Rufio87, 2017-08-24:
with chown you change the owner and by adding :www-data you change the group which owns the file or folder. with chmod g=rwx we give the full access to the group which owns the file folder. 

If you change owner group of all your files and folders created with samba from pi to www-data and enable rwx (read, write, execute) permission for the owner group to the same files and folders you should. You need to change these things for ALL files created with samba. There are some special commands with which you can change rights of all files in folders and subfolders very easily.

nachoparker, 2017-08-24:
wow @Rufio87, many thanks for the support!

It's a great help ;)

fab1ton, 2017-08-24:
I was indeed able to change the permissions but as you already mentioned I have to change it for each folder, which is a lot of work. Do you know the fast way how I can change it faster?

Rufio87, 2017-08-24:
@fab1ton 
you will find help here:
https://askubuntu.com/questions/693418/use-chown-to-set-the-ownership-of-all-a-folders-subfolders-and-files
https://stackoverflow.com/questions/3740152/how-do-i-set-chmod-for-a-folder-and-all-of-its-subfolders-and-files-in-linux-ubu

@nachoparker 
I thank you for this great project! 
what is also important: after the creation of the samba folder, specified in smb.conf as path, it would help changing the owner of the group to www-data and the group permissions to rwx. otherwise you cant for example delete files in the samba directory with a Nextcloud client.
Still, files added to Nextcloud with a Nextcloud client will not be fully accessible in the NAS created with samba (the files will only be readable).

nachoparker, 2017-08-29:
> I thank you for this great project!
what is also important: after the creation of the samba folder, specified in smb.conf as path, it would help changing the owner of the group to www-data and the group permissions to rwx. otherwise you cant for example delete files in the samba directory with a Nextcloud client.
Still, files added to Nextcloud with a Nextcloud client will not be fully accessible in the NAS created with samba (the files will only be readable).

Thanks, I know, I do something like that at my house but I don't know if it's a good enough solution to upstream it

nachoparker, 2017-08-29:
@Rufio87 

Hi, I got something better, for `v0.24.4`

```
force group = www-data
    create mask = 0770
    directory mask = 0771
    force create mode = 0660
    force directory mode = 0770
```

There you go, now you can create, read and write from both sides: samba and nextcloud :)

Thanks again for the pointers!

closed 2017-08-29 by nachoparker

nachoparker, 2017-08-29:
just so you know guys, I disabled SMB1 for security reasons (wannacry), so now the mount options will be different.

I'll update my blog post tomorrow with the info