nextcloud/nextcloudpi #132: ncp-web: password prompt for authentication

nachoparker, 2017-08-20:
Maybe using php-pam?

This would enhance security and also prevent from attacks even if the attacker obtains the cookie

testbird, 2017-09-17:
Ah, thanks for pointing me to the correct issue item.

The top solution from
https://serverfault.com/questions/45278/authenticate-in-apache-via-system-account
with the current syntax from the comments and adjusted to "require user pi", should do it. (Browser asking user for the credentials.)

Do you have a pointer to how such ncp updates are rolled out?

nachoparker, 2017-09-17:
cool, thanks very much!!

That solution seems like it would do at least for now. I will point the person that offered to help with this to this thread to see if he has also investigated and what his opinion is on this.

Maybe he has already implemented that same thing.

> Do you have a pointer to how such ncp updates are rolled out?

What do you want to know exactly? NCP checks for newer versions, and then updates them by executing [update.sh](https://github.com/nextcloud/nextcloudpi/blob/master/update.sh)

testbird, 2017-09-17:
Thanks, very god pointers to get an idea. I don't have enough background yet to write the update snippets, though.

Anyway, here is what configured a password box to pop up for me, now:
```
apt-get install libapache2-mod-authnz-external
a2enmod authnz_external

```

modified /etc/apache2/sites-available/ncp.conf:
```
Listen 4443
<VirtualHost _default_:4443>
  DocumentRoot /var/www/ncp-web
  SSLEngine on
  SSLCertificateFile      /etc/ssl/certs/ssl-cert-snakeoil.pem
  SSLCertificateKeyFile /etc/ssl/private/ssl-cert-snakeoil.key

  <IfModule mod_authnz_external.c>
    DefineExternalAuth pwauth pipe /usr/sbin/pwauth
  </IfModule>

</VirtualHost>

<Directory /var/www/ncp-web/>

  AuthType Basic
  AuthName "Login"
  AuthBasicProvider external
  AuthExternal pwauth

  <RequireAll>

   <RequireAny>
      Require host localhost
      Require local
      Require ip 192.168
      Require ip 10
   </RequireAny>

   Require user pi

  </RequireAll>

</Directory>
```
and a
`apachectl restart`

-----

PS: not sure now if any of these modules really had to be enabled manually
```
a2enmod authn_base
a2enmod authn_core
```

testbird, 2017-09-17:
At least with firefox, after the panel was idle for longer, the credentials were requested again when attempting to use the page again. But the javascript request did not seem to pick up the credentials (or new salt?). Completely reloading the page solved the issue.

nachoparker, 2017-09-18:
That's great!!

Would you like to submit a PR? we can polish the little details over it and I can write the update snippets. That way it will show up as your contribution

Thank you very much :)

nachoparker, 2017-09-18:
Also, luis can have a look at it and see what he thinks

lsousa, 2017-09-19:
Hi,

I haven't came across this solution before, but after reading a bit it really seems to be the best one to enable PAM auth.
I haven't yet tested, but I could try to catch errors on JS requests and if the request is forbidden, ask to reload page.

nachoparker, 2017-09-19:
@testbird 

> At least with firefox, after the panel was idle for longer, the credentials were requested again when attempting to use the page again. But the javascript request did not seem to pick up the credentials (or new salt?). Completely reloading the page solved the issue.

If you got 'unauthorized request' that's normal because the PHP session expired. One thing that needs to be improved is the handling of that situation ( at the moment tells you to refresh ).

Does this solution use PAM? I mean, do you need to use the system _pi_ password, which is _raspberry_ by default? 

@lsousa Thanks for offering help.

> I haven't yet tested, but I could try to catch errors on JS requests and if the request is forbidden, ask to reload page

That is something that is not finished in ncp-web and should be fixed. When that message appears sometimes refreshing is not enough, and you need to open a new tab. I did not spend time trying to debug that.

I think what we need to do is to see why that happens exactly and properly improve the security mechanism (which uses single use antiCSRF tokens). It is unrelated to this password issue, but something that needs to be fixed ;)

nachoparker, 2017-09-19:
also guys, when this feature is done, it will be very handy to implement this other one 

#130 

it is really easy, but it needed some extra security ;)

testbird, 2017-09-19:
Ok, makes sense: The timeout/reloding error I got was the unrelated php session timeout.

Yes, the solution uses PAM and the credentials of system user "pi" are needed to log in.

nachoparker, 2017-09-24:
implemented `v0.29.3`

closed 2017-09-24 by nachoparker