nextcloud/nextcloudpi #133: Wifi

swr85260, 2017-08-20:
I’ve got headache with the wi-fi connection. After a week of August vaccation I lost the connection, Weired, I decided to start form the new image file.
1/ wifi is not reconized and I need to update the Pi to activate it (I have a chinese pi board, not sure this is important to know)
2/ then, I use to make the wifi setup through the wpa_supplicant.conf file and it sounds stange and does not work any more (I tried differents ways of setup : wpa_supplicant or nc-wifi). Did you change something on the latest version?

nachoparker, 2017-08-20:
hi, 

let's see, what do you see on nc-wifi?

swr85260, 2017-08-20:
I started form scratch, to show you evidence.
Find the screen copy enclosed with only update on rev 0.24.0

----- Mail original -----

nachoparker, 2017-08-20:
mmm I don't think email attachments to github show up here, you might have to use the github web and drag the screenshot

swr85260, 2017-08-20:
[image]
[image]
this is the files

nachoparker, 2017-08-20:
raspberry pi 3?

swr85260, 2017-08-20:
Pi 3 yes

nachoparker, 2017-08-21:
And it works with the old image and not with the new one? even if you go back and forth between the two?

I have not changed anything, I don't think. Your wifi radio is not even detected.

Post the output of `dmesg`

swr85260, 2017-08-22:
[image]
Works fine with 0.24.1

closed 2017-08-22 by swr85260

nachoparker, 2017-08-22:
great!