nextcloud/nextcloudpi #135: Problem of login....

acavalli67, 2017-08-20:
Hi my name is Andrea,
I've faced this new problem if I try to login form the Lan.
I don't have any problem to log from external connection.....
That's the message:

You are logging on to the server from an untrusted domain.

Contact your system administrator. If you are an administrator of this instance, configure the "trusted_domains" setting in config / config.php. A sample configuration is available in config / config.sample.php.
Depending on your configuration, you can also use the bottom button to make this domain trusted as an administrator.

Pls help me.
Many thanks.

nachoparker, 2017-08-20:
How are you accessing? ip address? domain name?

acavalli67, 2017-08-20:
Ip address

nachoparker, 2017-08-20:
can you post the output of `ip address`?

acavalli67, 2017-08-20:
I'm so sorry but I don't know how I've to do, it's the first time that I'm using the Raspberry.... :-))

nachoparker, 2017-08-20:
no worries ^^

have you tried again?

do you have monitor and keyboard connected to the pi?

acavalli67, 2017-08-20:
yes

nachoparker, 2017-08-20:
ok, use password `raspberry` and user `pi` to log in, and then tell me the output of this command

```
ip address | grep glo
```

you have tried to connect again, right? it can take some seconds the first time you turn it on

acavalli67, 2017-08-20:
inet 192.168.1.111/24 brd 192.168.1.255 scope global eth0
inet 192.168.1.103/24 brd 192.168.1.255 scope global secondary eth0

nachoparker, 2017-08-20:
have you done any network configuration at all?

acavalli67, 2017-08-20:
Solved.... was incorrect the script for the network configuration of the fixed IP of Raspberry.....
Many thanks again.
Andrea

nachoparker, 2017-08-20:
no problem ;)

closed 2017-08-20 by nachoparker