nextcloud/nextcloudpi #137: fail2ban not working any longer after moving to Stretch

GHoeher, 2017-08-21:
Hi,

I upgraded my existing NextcloudPi installation to Stretch. After the upgrade fail2ban is not working any longer.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v 0.24.0

## What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
Not sure but it was the NextcloudPi delivered with 0.18.x. If it's possible to see it somewhere, I'll deliver this info also ...

### Expected behavior
fail2ban runs like before the upgrade

### Actual behaviour
fail2ban service fails to start (see error in log below).
Already removed and reinstalled fail2ban but error still exists.
Also already disabled and re-enabled fail2ban via nextcloudpi-config without success.
### Steps to reproduce, from a freshly installed image
upgrade to stretch via: 
sed -i 1d /etc/apt/sources.list
sudo apt-get dist-upgrade
sudo apt-get update

### Include logs
Aug 21 13:40:31 raspberrypi systemd[1]: Starting Fail2Ban Service...
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING 'backend' not defined in 'ssh'. Using default one: 'auto'
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING 'backend' not defined in 'nextcloud'. Using default one: 'auto'
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING 'filter' not defined in 'sshd'. Using default one: ''
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING No filter set for jail sshd
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING 'backend' not defined in 'sshd'. Using default one: 'auto'
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: WARNING 'filter' not defined in 'sshd'. Using default one: ''
Aug 21 13:40:31 raspberrypi fail2ban-client[10867]: ERROR  Failed during configuration: Bad value substitution: option 'action' in section 'sshd' contains an interpolation key 'port' which is not a valid option name. Raw value: '%(action_)s'
Aug 21 13:40:31 raspberrypi systemd[1]: fail2ban.service: Control process exited, code=exited status=255
Aug 21 13:40:31 raspberrypi systemd[1]: Failed to start Fail2Ban Service.
Aug 21 13:40:31 raspberrypi systemd[1]: fail2ban.service: Unit entered failed state.
Aug 21 13:40:31 raspberrypi systemd[1]: fail2ban.service: Failed with result 'exit-code'.
Aug 21 13:40:32 raspberrypi systemd[1]: fail2ban.service: Service hold-off time over, scheduling restart.
Aug 21 13:40:32 raspberrypi systemd[1]: Stopped Fail2Ban Service.
Aug 21 13:40:32 raspberrypi systemd[1]: fail2ban.service: Start request repeated too quickly.
Aug 21 13:40:32 raspberrypi systemd[1]: Failed to start Fail2Ban Service.
Aug 21 13:40:32 raspberrypi systemd[1]: fail2ban.service: Unit entered failed state.
Aug 21 13:40:32 raspberrypi systemd[1]: fail2ban.service: Failed with result 'exit-code'.

nachoparker, 2017-08-21:
Thanks for reporting. I didn't notice that in my testing.

I'm a little bit busy now, so if anybody can help debug what is going on with the jails, it should be easy. Probably the new version of fail2ban has different syntax for its config files.

Also it would be nice to include this kind of thing for automated testing at the end of the build.

jb8628, 2017-08-21:
I found that /etc/fail2ban/jail.d/defaults-debian.conf contains a [sshd] entry for which no port (or anything else) is defined. This entry either needs to be completed or disabled/removed.

Now working for me.

nachoparker, 2017-08-21:
Great!

@jb8628 so, what's the output of

```
sudo fail2ban-client status sshd
```

jb8628, 2017-08-21:
```Status for the jail: ssh
|- Filter
|  |- Currently failed:	0
|  |- Total failed:	0
|  `- File list:	/var/log/auth.log
`- Actions
   |- Currently banned:	0
   |- Total banned:	0
   `- Banned IP list:	
```

Works because [ssh] is in the jail.conf file, nothing to do with the incomplete [sshd] entry.

nachoparker, 2017-08-21:
Well, both jails look ok then

Would you like to submit a pull request? otherwise I'll change myself

jb8628, 2017-08-21:
Sorry, I'm new to all this and have no time to figure out pull requests tonight before leaving on holiday for a couple of weeks. Next time!

It's a tiny fix anyway. I just commented out the content of /etc/fail2ban/jail.d/defaults-debian.conf.

Thanks for all your efforts on NextCloudPi. Really appreciated and you've inspired a lot of learning!

nachoparker, 2017-08-21:
Cool, thanks again for the help

If you feel like contributing again in the future, all help is welcome

Have fun!

nachoparker, 2017-08-21:
fixed `v0.24.1`

closed 2017-08-21 by nachoparker