nextcloud/nextcloudpi #138: Nc-backup and nc-restore: question

Ricardosgeral, 2017-08-21:
Hi all

Let's say I have done a nc-backup to an external drive where I also have the data. 

If I Install a new fresh image of nextcloudpi in a new SD card, then what are the procedures for restoring the database? 

Is just a matter of doing nc-restore? 
It's not necessary to do the following sequence? 
nc-automount
nc-init
no-ip
port forwarding 
Letsencript
... 
Thanks

### What version of NextCloudPi are you using? ( eg: v0.17.2 )

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

### Expected behavior

### Actual behaviour

### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-08-21:
Hi, 

At this moment, doing nc-restore will restore your Nextcloud, including database and also the datafiles if you selected that option at the time of backup

The configuration of NextCloudPi will NOT be stored there, so you will have to do those configuration steps again.

Actually that has been on my head for some time but I haven't prioritized it: the ability to save/restore NCP configuration. I'll mark this as an enhancement

Ricardosgeral, 2017-08-21:
The database includes the user's permissions.  That is, can I use the data files in the external files or will I have to upload every file again. 
Should I do nc-restore first and then the other steps or at the end?

nachoparker, 2017-08-21:
You should do the same steps in the same order, with the possible exception of `nc-datadir`

 - automount
 - DDNS
 - dnsmasq
 - Let's Encrypt

The order of the rest of the things ( swap, fail2ban... ) does not matter.

In case you included data in your backup, you don't have to do `nc-datadir` again. In case you didn't you will have to either upload them again or copy them in Linux to the right folder and then run `nc-scan`

closed 2017-08-21 by nachoparker