nextcloud/nextcloudpi #139: headline of ncp-web version xx.yy.zz pointing to changelog?

JimmyKater, 2017-08-21:
see subject box... is this possible?
would be just another enhancement, i think (btw how could i put this label on all on myself?)

nachoparker, 2017-08-21:
Hi Jimmy!

> see subject box... is this possible?

Do you mean click the version number and it takes you to the changelog?

> btw how could i put this label on all on myself?

What do you mean?

JimmyKater, 2017-08-21:
> Do you mean click the version number and it takes you to the changelog?

exactly.

> What do you mean?

i mean... this if off-topic... i see you are self-assigning all of the issues that come up here and then give them a label. if i would know how to do it myself i'd put the label "enhancement" to this one here... ;)

nachoparker, 2017-08-21:
> exactly.

Got it. I like the idea... let's add it to the wishlist! We would have to first generate and store some kind of changelog... but that would certainly something nice to have

JimmyKater, 2017-08-21:
ummm since it's your project of course youre allowed to self-assign stuff. that wasn't my point. i just wanted to spare you some clicks and add the label(s) myself. i'm just too noob, still to know how to do it myself. :blush:

nachoparker, 2017-08-21:
That'd be great, like I said, I'll study how that works when I have time. 

We gotta get the contribution system going so that anybody can help!

nachoparker, 2017-09-07:
Ok, now the changelog is generated automatically with each build

https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md

Now we only need a link ;)

nachoparker, 2017-09-07:
implemented `v0.26.17`

closed 2017-09-07 by nachoparker

JimmyKater, 2017-09-08:
ummm. do i need to restart or does it only work for stetch? b/c i just updated (to 26.20) and find it didnt worked.
neither the email setup.

nachoparker, 2017-09-08:
lol, I just read this again. I made a clickable link pointing to changelog, but only in the "an update is available" message. Not on the current version number.

Now I see that's what you meant. Mmmm yes, I guess that's a better place. It is a bit hidden, maybe we should make it more explicit.

Email didn't work? Let's see if it is installed

```
type sendmail
```

How did you test it?

JimmyKater, 2017-09-08:
i just found out how you made email work in a different topic. maybe we should make it a new "issue"?

JimmyKater, 2017-09-08:
>  I made a clickable link pointing to changelog, but only in the "an update is available" message. Not on the current version number.

Now I see that's what you meant. Mmmm yes, I guess that's a better place.

why not making both? since it does make sense to see the changes before you do the update

nachoparker, 2017-09-08:
> maybe we should make it a new "issue"?

do you mean a "how to use email" entry? like in the wiki or something like that?

> why not making both?

actually... yeah! :D

JimmyKater, 2017-09-08:
strange, my comment vanished. so i''m gonna try remembering it again: 

> do you mean a "how to use email" entry? like in the wiki or something like that?

at least i think you should tell every user that you're setting up email for them (and what you need to make it work). this would be without any problem for first users(/installations. 
but as some of us are updating ncp and thus are existing users there should be a routine to make sure that your defaults doesn't overwrite an existing/working email-setup.

nachoparker, 2017-09-09:
Well, the update only installs and configures stuff if there is nothing configured, so it should not override stuff ;)

Thanks for the warning, well thought!

JimmyKater, 2017-09-09:
>  it should not override stuff

but it did with my settings. no biggie, though... but i was wondering why it got changed. and when i read about the implementation of email support it got clearer to my...

JimmyKater, 2017-09-10:
>  it should not override stuff

but it did with my settings. no biggie, though... but i was wondering why it got changed. and when i read about the implementation of email support it got clearer to me...

nachoparker, 2017-09-10:
Ok, new stuff

- the update box takes you to `nc-update`
- the version takes you to changelog
- there's also links to the wiki for each extra
- the info box from `nextcloudpi-config` shows also on the web now

:)

nachoparker, 2017-09-10:
[image]

nachoparker, 2017-09-10:
I'll use this to be able to upload a picture to the wiki

[image]