nextcloud/nextcloudpi #141: First configuration steps

niconoob, 2017-08-21:
Hi, very basic question here. After the pi is up an running with ncp, the next step would be to configure nextcloud itself. This can be done via the web interface. I've read in ownyourbits about the "admin" and "ownyourbits" defaults, but nothing about the fields that are required below that for the database (database user, database password and database name, also localhost+portnumber). If I ignore them, nextcloud complains, but I haven't found any combination that works (e.g. repeating the nc admin defaults didn't work). 
In earlier attempts to set up nc on my own, I had to configure mysql before. Does that still apply here? I thought ncp took care of that.
What am I doing wong? Can you point me to more detailed "first steps" resource?

nachoparker, 2017-08-21:
Hi,

Nextcloud is already up and running and preconfigured for you from the moment you plug in the Raspberry Pi.

Did you run `nc-nextcloud`? Just so you know, you don't need to run `nc-nextcloud` or `nc-init`. Those are just for reinstalling a different version and resetting the configuration respectively.

niconoob, 2017-08-22:
Hi,

Thanks for the reply. I don't remember having run nc-nextcloud, though I
definitely have run nc-init at least twice trying to get things up and
running.
Can i "undo" nc-nextcloud or whatever I've messed up or am I better off
with flashing the new image and starting from scratch?

nachoparker, 2017-08-22:
Well, probably the easiest is to run `nc-nextcloud` followed by `nc-init`

That wipes out everything you had in nextcloud ( configs, users, files if not moved datadir ), but it seems like you don't have anything yet.

closed 2017-08-22 by nachoparker