nextcloud/nextcloudpi #142: Automount creates new mountpoint at every reboot

GHoeher, 2017-08-22:
### PROBLEMS

Since version 0.24.0, NextcloudPi creates a new folder/mountpoint in the media folder at every reboot.
The USB-Stick gets mounted to "randomly" to folder /media/USBdriveX until USBdrive8-Folder exists.
Afterwards, it always gets mounted to /media/UBSdrive.
You need 6 reboots to get a stable mounting of the USB-Stick.
I'm using only one USB-Stick in my configuration.

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
0.24.1

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_07-29-17 upgraded to Stretch and now v0.24.1
### Expected behavior
USB-Stick gets mounted to the same folder at each reboot

### Actual behaviour
folder /media/USBdrive exists.

Reboot -> USBdrive1 and USBdrive2 created and Stick mounted to USBdrive1 and USBdrive2
Reboot -> USBdrive3 and USBdrive4 created and Stick mounted to USBdrive3 and USBdrive4
Reboot -> USBdrive5 and USBdrive6 created and Stick mounted to USBdrive5 and USBdrive6
Reboot -> USBdrive7 created and Stick mounted to USBdrive and USBdrive7
Reboot -> USBdrive8 created and Stick mounted to USBdrive8
Reboot -> no new folder and and Stick mounted to USBdrive 
Reboot -> no further change

### Steps to reproduce, from a freshly installed image

### Include logs
fstab after the complete "cycle":
proc            /proc           proc    defaults          0       0
PARTUUID=f5d16add-01  /boot           vfat    defaults          0       2
PARTUUID=f5d16add-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

# Rules for automounting both at boot and upon USB plugin. Rely on udev rules
/dev/USBdrive  /media/USBdrive         auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive1 /media/USBdrive1        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive2 /media/USBdrive2        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive3 /media/USBdrive3        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive4 /media/USBdrive4        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive5 /media/USBdrive5        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive6 /media/USBdrive6        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive7 /media/USBdrive7        auto    defaults,noatime,auto,nofail    0       2
/dev/USBdrive8 /media/USBdrive8        auto    defaults,noatime,auto,nofail    0       2
tmpfs /var/log tmpfs defaults,noatime,mode=1777 0 0 # Logs in RAM
tmpfs /tmp     tmpfs defaults,noatime,mode=1777 0 0 # /tmp in RAM

nachoparker, 2017-08-22:
How is the USBdrive formatted?  output of 

```
lsblk -f
```
?

GHoeher, 2017-08-22:
Output of lsblk is:

NAME        FSTYPE LABEL       UUID                                 MOUNTPOINT
sda
└─sda1      vfat   USBDrive8GB 34D8-6159                            /media/USBdrive
mmcblk0
├─mmcblk0p1 vfat   boot        109A-9113                            /boot
└─mmcblk0p2 ext4               673b8ab6-6426-474b-87d3-71bff0fcebc3 /

nachoparker, 2017-08-22:
`vfat` is not supported by Nextcloud, even though it is supported by automount. Last time I checked it was working.

Did this work for you before moving to Stretch? did you move to Stretch? Manually, or new image?

GHoeher, 2017-08-22:
It worked before moving to Stretch (on Jessie there only was USBdrive and USBdrive1 and the stick always got mounted to USBdrive1). I moved to Stretch manually.
What exactly does mean "vfat is not supported" by Nextcloud? I don't use it directly for nextcloud data, but only for the backups, so this should be ok in my eyes ...

nachoparker, 2017-08-22:
> What exactly does mean "vfat is not supported" by Nextcloud? I don't use it directly for nextcloud data, but only for the backups, so this should be ok in my eyes ...

It means that you cannot move the datadir or database to a `vfat` partition, but for backups it's ok ;)

This is because that filesystem does not support Unix permissions, which are used by the database and Nextcloud for security

I will test this again on stretch

Thanks for reporting this!

deljones, 2017-08-24:
Yes same here.. I'm up to USBdrive8 now... Whew.. That's happened some evening upgrading to 0.24.1

System is working fine though, I think, otherwise 

Dj

nachoparker, 2017-08-24:
@deljones thanks for confirmation.

Seems like systemd changed quite a bit its udev behaviour in Stretch.

At least it works for now. Unfortunately, the fix will have to wait, I'm off to the Nextcloud conference in Berlin ;)

deljones, 2017-08-24:
No worries.. Have a good time in Berlin 

Dj

w: derekjones.net

e: derek AT derekjones.net

t: +44 7957 140 512

GHoeher, 2017-08-24:
@deljones: If it behaves the same way like my installation, after the next reboot it should mount the USB-Stick to USBdrive and then you have a stable configuration no matter how often you reboot the RaspPi.

@nachoparker: for me it's also no problem if this takes some more time to fix this bug as I'm currently running stable after 6 or 7 reboots :-) I also wish you a good time in Berlin!

nachoparker, 2017-08-24:
thanks guys :D

nachoparker, 2017-08-30:
fixed `v0.24.7`

nachoparker, 2017-08-31:
The automount feature was rewritten from scratch due to it being increasingly incompatible with systemd-udev. It got to a point where it just didn't work in Stretch.

The new implementation also supports encrypted disks and several partitions per drive.

Implemented `v0.24.16`

closed 2017-08-31 by nachoparker